gilestrolab/ethoscope #5: Update system reports error when everything is OK

qgeissmann, 2015-08-20:
_From @qgeissmann on April 14, 2015 10:13_

When a new version is available, we always get errors such as

```
Error details: POST git-upload-pack (959 bytes) From https://github.com/gilestrolab/pySolo-Video e9b8129..9a859c9 psv-dev -> psv-dev
```
Simply ignoring the error and proceed to update works perfectly though.

_Copied from original issue: gilestrolab/pySolo-Video#95_

qgeissmann, 2015-08-20:
_From @PolygonalTree on April 14, 2015 10:42_

It is a bug on Git:
http://stackoverflow.com/questions/10790232/hanging-at-post-git-receive-pack-chunked

qgeissmann, 2015-09-14:
I would like to introduce the idea of **using cron jobs to pre-fetch updates*.

* Every 5min, node bare repo would fetch the last version of `dev` **and `master`.
* Every 5min, node working repo will fetch, but **not** pull from bare.
* Every 5min, devices working repos will fetch, but **not** pull from bare.

On update, devices (and node) are only only using local data to update.
This would help on several issues:

* the current one (#5)
* The fact that updates are slow (several minutes) when they are rare, as we fetch and pull a large tree every time instead of doing it incrementally.
* the fact that the user does not really know what is going on if it fails (e.g. connection hangs. problem in restarting the devices...?)
* it makes the server more stable as we delegate a task to another program.

What do you think @pepelisu

pepelisu, 2015-09-15:
No everyone will have the node connected to internet and therefore the cron job would be doing nothing. It should be an option somewhere in the configuration of the node. i.e under "More -> node config" tab.
Time between update checking should be also a parameter. 5 min is ok for dev but to much for production. We have to have in mind that we only merge in master once in a while and not regularly.

qgeissmann, 2015-09-15:
It is not a super high priority matter, but, for me:

When there is no internet connection, it is OK if the cron job does nothing. The user will not see anything, it won't impact performance, and it is not on the server, so it does not compromise stability.

When there is an internet connection, the cron job will just get an "everything is up to date" message back. which is fine too, even if it is every 5 minutes.

qgeissmann, 2015-09-18:
this is solved by #32 as now we use [git python API] (https://gitpython.readthedocs.org/en/stable/)

closed 2015-09-18 by qgeissmann