gilestrolab/ethoscope #6: Node crashes if connection stops whilst downloading large file

qgeissmann, 2015-08-20:
_From @qgeissmann on April 20, 2015 17:9_

Was downloading a large file. Connection was interrupted. could not connect to the node back
Related to #74

_Copied from original issue: gilestrolab/pySolo-Video#97_

qgeissmann, 2015-08-20:
_From @PolygonalTree on April 20, 2015 17:13_

Maybe related with this:
http://stackoverflow.com/questions/11866792/how-to-prevent-errno-32-broken-pipe

qgeissmann, 2015-08-20:
_From @PolygonalTree on April 20, 2015 17:14_

Here possible solution, catching the error on the download function. http://stackoverflow.com/questions/180095/how-to-handle-a-broken-pipe-sigpipe-in-python/180922#180922

qgeissmann, 2015-08-20:
_From @PolygonalTree on April 20, 2015 17:23_

Other possible solution, is Request library a urllib2 with steroids. http://stackoverflow.com/questions/16694907/how-to-download-large-file-in-python-with-requests-py

qgeissmann, 2015-08-20:
btw? you think, at the moment, the file is copied in memory before upload? if yes, this is a clear issue... e.g. i had a 700MB zip to dl for 4 days x 4 devices

qgeissmann, 2015-08-20:
_From @PolygonalTree on April 23, 2015 14:56_

I have made some test and I tried to download a file of 1.3Gb directly from Node. It worked with no errors. I also tried to kill the connection on the middle of the transaction and the transaction stops without problem, Node does not hang. 
I have also checked the memory during transaction and Node is not loading the file on memory before it serves it.
So not really sure where the problem is. Do you have more info?

PolygonalTree, 2015-09-02:
What is the status of this bug? wasn't it closed?

qgeissmann, 2015-09-02:
I think that, now that we have a completely independent backup daemon, we should not have this problem. closing.

closed 2015-09-02 by qgeissmann