gilestrolab/ethoscope #8: "start" POST should include a dict with additionnal information

qgeissmann, 2015-08-20:
_From @qgeissmann on May 27, 2015 14:51_

We are starting to have different arenas, so, I would like to be able to not only "start a device", but "start a specific arena". Long term, we could use bare codes, but I still think this is a needed feature.

_Copied from original issue: gilestrolab/pySolo-Video#106_

qgeissmann, 2015-08-20:
_From @PolygonalTree on July 14, 2015 20:9_

This has been addressed and a first implementation is available on dev but we have to discuss what additional information is required.

qgeissmann, 2015-08-24:
OK so I thought about how to implement that in a generic fashion (this should require only minor changes when new tracker, roi builder and interactors are implemented).
It requires some changes in both the front and back-end though, so we  (@PolygonalTree) need to agree on that.

 For clarity, I will describe it here quickly:

Global change in the back-end
-------------------------------------

The classes `ROIBuilder`, `Tracker` and `Interactor` will now derive from a new abstract class named `DescribedObject`. Which simply had a `__description` attribute. 
description is a dictionary which should look like this:

```python
__description = {
# The global description about the function of this class e.g.
    overview="Builds 20 regions of interests on two columns and ten rows. Uses the black visual target drawn on the arena",
    args=[
        # the first arg
        {    name: "the name of the arg",
             default_value: 2,
             arg_description: "a specific description for this argument",
            # We probably need to specify a type and range as well if we don;t want user to set crazy values
             {type:["float","int","str"], min: 0.0, max: 100}
        },
        # more user changeable arguments
        {},...
    ]
}
```
Changes in device server
-------------------------------------

The `ControlThread` will have an explicit dictionary listing all classes available for the user for `ROIBuilder`, `Tracker` and `Interactor`. This can be exposed publicly via bottle/POST, and sent as JSON.

This class should also have a JSON parser which:

* get classes
* checks arguments are in the right type and range (error otherwise)
* make instances to check argument validity ?

Front-end changes
--------------------------------------

Using the requested JSON, we can, on start, prompt a two panel dialog.

* the header allowing to select existing profile, or custom.
* the left panel has three drop down menus (`ROIBuilder`, `Tracker` and `Interactor`). It also allows to 'select' one  (see next, right panel)
* the right panel displays description of the selected menu and allow to select values for each argument.

The formatted JSON should  be : 

* saved in the metadata of the .db
* saved, under user request, locally (and under a user selected name). this allows to use it as a profile next time.

qgeissmann, 2015-08-25:
ok so we are almost there for the back-end.

in order to help front-end dev. (@PolygonalTree):

There is a new `user_options` field in `ControlTread._info`.

An example of formatted `user_options` could look like this(from the actual JSON):
```python
{"tracker":[
    {"overview":"The default tracker for fruit flies. One animal per ROI.",
     "arguments":[],
     "name":"AdaptiveBGModel"}],
 "roi_builder":[
     {"overview":"The default sleep monitor arena with ten rows of two tubes.",
      "arguments":[{"description":"The number of columns","min":1,"default":1,"max":64,"type":"int","name":"n_cols"},
                   {"description":"The number of rows","min":1,"default":1,"max":64,"type":"int","name":"n_rows"}],
      "name":"TargetArenaTest"}],
 "interactor":[
     {"overview":"The default sleep monitor arena with ten rows of two tubes.",
      "arguments":[],
      "name":"DefaultInteractor"}]
}
```
Note that these are automatically generated from the description implemented within each class (i.e. modular). 
You can see that arguments are optional. 
When they are present, their description, type and range should be sufficient to build a front end interface.

**The goal of the front-end is to format user options and start with a POST that looks like**:

```python
data = {"roi_builder":{"name":"TargetArenaTest",
                              "arguments":{"n_cols": 2, "n_rows": 4}},
                "tracker": {"name":"AdaptiveBGModel",
                              "arguments":{}},
                "interactor": {"name":"DefaultInteractor",
                              "arguments":{}},
        }
```
I think the dual panel approach I suggested in the previous post should work well for the user interface.
It would be quite nice to have the option to send a saved profile (simply a JSON) early on for development purposes.

PolygonalTree, 2015-08-31:
Structure:
`{description: "something useful", name:"name", type:"Type", min:"1", max:"100", step:"0.1", default:"1"}`
Lets define types:
1. number: float or int (defined by step).
2. str: characters
3. datetime: Internally unix timestamp
example:
{"tracker":[
    {"overview":"The default tracker for fruit flies. One animal per ROI.",
     "arguments":[],
     "name":"AdaptiveBGModel"}],
 "roi_builder":[
     {"overview":"The default sleep monitor arena with ten rows of two tubes.",
      "arguments":[{"description":"The number of columns","min":1,"default":1,"max":64,"type":"number", "step":1, "name":"n_cols"},
                   {"description":"The number of rows","min":1,"default":1,"max":64,"type":"number", "step":1, "name":"n_rows"}],
      "name":"TargetArenaTest"}],
 "interactor":[
     {"overview":"The default sleep monitor arena with ten rows of two tubes.",
      "arguments":[],
      "name":"DefaultInteractor"}]
}

qgeissmann, 2015-08-31:
OK, @PolygonalTree  I have updated `TargetArenaTest` consequently. Its description looks like:

```python
    description = {"overview": "The default sleep monitor arena with ten rows of two tubes.",
                    "arguments": [
                                    {"type": "number", "min": 1, "max": 64, "step":1, "name": "n_cols", "description": "The number of columns","default":1},
                                    {"type": "number", "min": 1, "max": 64, "step":1, "name": "n_rows", "description": "The number of rows","default":1},
                                    {"type": "str", "name": "dummy_str", "description": "The number of rows","default":"Luis' easter egg"},
                                    {"type": "number", "min": -2.1, "max": 5.2, "step":0.01, "name": "dummy_float", "description": "Another dummy parameter to test","default":0.1},
                                    {"type": "datetime", "name": "dummy_datetime", "description": "Can we set a date","default":1441035646}
                                   ]}
```
an exple JSON config file could be:

```js
{"roi_builder":{"name":"TargetArenaTest",
                      "arguments":{ "n_cols": 1,
                                    "n_rows": 1,
                                    "dummy_str":"yo man",
                                    "dummy_float":3.11,
                                    "dummy_datetime":1441036099
                                  }
                  },
        "tracker": {"name":"AdaptiveBGModel",
                      "arguments":{}},
        "interactor": {"name":"DefaultInteractor",
                      "arguments":{}}
}
```

runs with e.g. 

```
$python device_server.py -g  /home/quentin/comput/ethoscope-git/ -r -i /data/psv_misc/monitor_validation/monitor_validation_raw.mp4  -d -j ../../prototypes/json_config/config.json
```

qgeissmann, 2015-09-02:
I was just thinking, It would be nice to add another type: `src`.
This would be a special string to be **evaluated** (by the `python` interpreter) in the user option parser (in device). For instance it could be `"[i for i in range(10)]"`.
In the user interface, it is simply a multi-line text entry.
This is will be useful when we have, during development, to pass arguments such as containers, or more complicated objects.

pepelisu, 2015-09-02:
That can be useful for development, but it is really a black hole in terms of security, we can discuss this though.

qgeissmann, 2015-09-02:
I agree about the security. I allows to do virtually anything...

pepelisu, 2015-09-02:
We can use it in dev branch, but not in production (or master).

qgeissmann, 2015-09-02:
yes, let's do something like that, that we can simply deactivate, so we prevent users to break their own system (they probably don't need help for that ;)).

qgeissmann, 2015-09-03:
@pepelisu just to let you know a potential bug in the user interface:

When I change from the sleep depriver interactor to the `DefaultInteractor`, which has no arguments, the list of arguments seems to be taken from the  sleep depriver interactor, somehow. In the end, this sends an inconsistent POST:

```js
...
"interactor":{
     "name":"DefaultInteractor",
     "arguments":{
                 "velocity_threshold":0.006,               
                 "min_inactive_time":120,
                 "start_datetime":0,
                 "end_datetime":2147483647}
}
...
```

The expected behaviour is:

```js
...
"interactor":{
     "name":"DefaultInteractor",
     "arguments":{}
}
...
```

qgeissmann, 2015-09-04:
I am really happy with your implementation @pepelisu. This issue is a bit packed, so I close it, and we will open a new one specific for the few front end thing we need to do e.g. default config vs advanced (the one we have now is only advanced).

closed 2015-09-04 by qgeissmann