gilestrolab/ethoscope #16: Leap second breaks system

qgeissmann, 2015-09-01:
Looking in journalctl, we can see that a *leap second has been inserted*.
Immediately after, devices raise an exception:

```python        
dt = float(t - self.last_t)
if dt < 0:
    raise EthoscopeException("Negative time interval between two consecutive frames")
```

qgeissmann, 2015-09-01:
That is quite unfortunate:
from the [wiki](https://en.wikipedia.org/wiki/Leap_second) article" Since this system of correction was implemented in 1972, 26 leap seconds have been inserted, the most recent on June 30, 2015 at 23:59:60 UTC.[1]"

qgeissmann, 2015-09-01:
There is quite a bit of mystery for me here.

1. Why has leap second been inserted in 2015-08-31 23:59:60? Leap seconds are meant to be in the last days of June and December.
2. Why do the above code lead to negative time? At worst, we should get `dt==0`.

ggilestro, 2015-09-01:
Are you sure this is not simply due to NTP adjusting time? perhaps the
monitor was running for too long and time drifted considerably

qgeissmann, 2015-09-01:
It seems that the kernel itself changed time with an explicit leap second:
```
Aug 31 23:59:59 00016dfce6e94dee9bb1a845281b086e kernel: Clock: inserting leap second 23:59:60 UTC
Aug 31 23:59:59 00016dfce6e94dee9bb1a845281b086e systemd[1]: Time has been changed
Aug 31 23:59:59 00016dfce6e94dee9bb1a845281b086e python2[2983]: ERROR:root:Traceback (most recent call last):
Aug 31 23:59:59 00016dfce6e94dee9bb1a845281b086e python2[2983]: File "/home/psv/pySolo-Video/src/pysolovideo/web_utils/control_thread.py", line 170, in run
Aug 31 23:59:59 00016dfce6e94dee9bb1a845281b086e python2[2983]: self._monit.run(rw)
Aug 31 23:59:59 00016dfce6e94dee9bb1a845281b086e python2[2983]: File "/home/psv/pySolo-Video/src/pysolovideo/tracking/monitor.py", line 224, in run
Aug 31 23:59:59 00016dfce6e94dee9bb1a845281b086e python2[2983]: raise e
Aug 31 23:59:59 00016dfce6e94dee9bb1a845281b086e python2[2983]: PSVException: 'Negative time interval between two consecutive frames'
Sep 01 00:15:09 00016dfce6e94dee9bb1a845281b086e ntpd[382]: kernel reports leap second has occurred
Sep 01 00:15:09 00016dfce6e94dee9bb1a845281b086e ntpd[382]: kernel reports leap second has occurred
Sep 01 01:59:40 00016dfce6e94dee9bb1a845281b086e systemd[1]: Time has been changed
```

qgeissmann, 2015-09-01:
I don't think it is a simple time drift issue as **all** devices that were tracking crashed at the exact same time.
This look more like a kernel 'bug' to be.

PolygonalTree, 2015-09-01:
Negative times are due to the difference in microseconds.
I don't get why the kernel should change the time in devices. It is provided and managed by NTP service.

commit c8aa7543 (2015-09-14): patch. fixes #16
closed 2015-09-14 by qgeissmann