gilestrolab/ethoscope #18: Arena targets holes should be square

estebanbeck, 2015-09-02:
To have better recongnition of the targets would be easy if the printed holes are square. Then we cut the paper with the printed dots in squares (easier than cut rounded shapes) and fit them in the holes.

qgeissmann, 2015-09-02:
I agree. At the moment, targets are much better when printed on paper first, and the glues on arena. This would help us to ensure they are at the right place (which is hard when trying to glue a disc inside a disc).

PolygonalTree, 2015-09-02:
OK square they will be (in the future), but please software need to be 
compatible with rounded ones.

qgeissmann, 2015-09-02:
@PolygonalTree  the holes only should be square, but the target is still visually similar (circle).
The idea is that is is much easier to stick a peace of paper that is circular inside a square.
The side of the square should obviously be the same as the diameter of the actual target.

qgeissmann, 2015-09-02:
Also, @PolygonalTree. Any chance we can actually have, by convetion, the distance between the edge of the A and the top left tube the same as the distance between B and the bottom left tube:

```
        #                            A
        #                            |
        #                            |
        #                            |
        # C------------------------- B
```

You see what I mean?
This would make things easier (see #20 also).

PolygonalTree, 2015-09-02:
new Arena dimensions. Check target distance and let me know if this could work.
[image]

qgeissmann, 2015-09-02:
Just to be sure,  `x == y` right?
[image]

PolygonalTree, 2015-09-02:
yes sure.
Distance from centre of the square to centre of first tube is 6.99mm in both sides (now everything is centred).

qgeissmann, 2015-09-02:
Great.
@PolygonalTree  can you confirm to me these other measures:

- [x] distance between centre of square and centre of closest tube:  6.99mm
- [ ]  distance between two consecutive tubes
- [ ]  distance |AB| (centre to centre)
- [ ]  distance |BC| 

cheers

pepelisu, 2015-09-02:
Distance between two tubes = 6.4mm
Distance AB = 71,58mm
Distance BC = 111mm

closed 2015-09-03 by qgeissmann