gilestrolab/ethoscope #22: tests

qgeissmann, 2015-09-02:
how that we have command line interface for the device sever and config file, we could have a list of config files that we run so we test software on video files (instead of live stream) at each push. Of course, interactors that involve using external hardware cannot be tested this way, but is would still be good to have.

pepelisu, 2015-09-02:
Agree.

qgeissmann, 2015-09-04:
Now, we have a very generic JSON file configuration, where we can input almost all classes used in Monitor with specific arguments. This means we can write a dir containing config files `*.json` and run:
` python device_server.py -g  /home/quentin/comput/ethoscope-git/   -j my_file.json -r`
on it.

qgeissmann, 2015-09-18:
Some integrations tests are done. More would be better, but closing for now

closed 2015-09-18 by qgeissmann

qgeissmann, 2015-09-23:
An update on that. I have just made another integration test to check whether the device server responds as expected.
It basically starts a the bottle server in the background. then, it uses `curl` to POST/GET to start/stop/close the tracking tool.
This allows to check we did not brake anything in the way tracking is controlled.

Please, run it with:

```
cd src/ethoscope/tests/intergation_server_tests
sh test_post.sh
```