gilestrolab/ethoscope #25: Name the experiments

estebanbeck, 2015-09-04:
Could be possible to give a name to the experiment when we start the ethoscopes? If this is a metadata then we can retrieve it in R.

qgeissmann, 2015-09-04:
Good idea, this is now quite easy to do, with user options interface. then we can simply link the "experiment_name" in the metadata.

qgeissmann, 2015-11-30:
Ok, so I co-opted the option parsing mechanism to get simply this:

[image]

Also we display it in the device list table

[image]

still need to make the table sortable (see #40)

closed 2015-11-30 by qgeissmann