gilestrolab/ethoscope #32: Update system should be a separate daemon.

qgeissmann, 2015-09-18:
There is quite a lot of instability in the current update system due to the fact that the current update system is part of the bottle server.

Three main criticisms have been addressed:

1. When updating, we update the updating system, which can break it.
2. If bottle server are broken for any reason (e.g. related to tracking/network/...), they cannot be updated.
3. Once finished, the update system is responsible of closing its own process, forcing systemd to restart it, which is quite dirty.

As a result, we have started designing a new update system that will be be a minimalist server running on port 8888 (maybe we can make it available at http://ethoscope_update/).

The back-end is actually ready ( feature branch => https://github.com/gilestrolab/ethoscope/tree/fix_update), so we just need to code the JS/HTML front end.

In addition to simply allow to go to update to latest versions, the new update system allows to switch to detect new branches upstream, and switch devices/node to new branches. This feature will be very useful to develop features without breaking the stable branch.

qgeissmann, 2015-09-21:
@pepelisu, I sketched a front end, which I enclosed. After the figure, I described, in pseudocode, how actions should be handled. Finally, I explained which headers the table should have.

To my opinion, it should be a single web page  with a table resembling a lot the one we had in  the former update tab:

[image]

Actions
===========

on load page
-----------------
```
fetch upstream [API ref => node_ip:8888/bare/update]
load device_map  [API ref => node_ip:8888/devices]
get node version  [API ref => node_ip:8888/device/check_update]
get local ref branch  [API ref => node_ip:8888/device/active_branch] # node active branch is the ref

for each device d:
    get device version  [API ref => d_ip:8888/device/check_update]
    get dev active branch  [API ref => d_ip:8888/device/active_branch]
    add device and version in device_map

display table
```

on update
---------------

```
Get selection (contains devices AND node)
RE-check devices are not "running"
Show modal window "you are about to update N devices"
if proceed:
    for each d in selection:
         change branch -> reference branch   [API ref => d_ip:8888/device/change_branch]
         fetch/pull [API ref => d_ip:8888/device/update]
         restart daemon  [API ref => d_ip:8888/device/restart_daemon] # todo
    reload page
```

on restart
---------------

```
Get selection (contains devices AND node)
RE-check devices are not "running"
Show modal window "you are about to restart software on N devices"
if proceed:
    for each d in selection:
         restart daemon  [API ref => d_ip:8888/device/restart_daemon] # todo
    reload page
```

on change branch
---------------

```
Get selection (contains devices AND node)
Get available branches [API ref => node_ip:8888/device/available_branches]
RE-check devices are not "running"
Show modal window "N devices selected... which branch would you like to switch to" -> drop-down menu

if proceed:
    change reference branch   [API ref => node_ip:8888/device/change_branch]
    for each d in selection:
         change branch -> new reference branch   [API ref => d_ip:8888/device/change_branch]
         fetch/pull [API ref => d_ip:8888/device/update]
         restart daemon  [API ref => d_ip:8888/device/restart_daemon] # todo
    reload page
```

Table structure
============

Node should always be the first row.

Headers:

* Name (e.g. Node, DEVICE_001,...)
* Status (for node: {OK, software broken} for device: {running, stopped, software broken})
* Software version ( format= branch, commit_id (commit date))
* ip
* id
* **select**

qgeissmann, 2015-09-23:
In my previous post I explain that we need to restart daemons after update.

I proposed:
```
restart daemon  [API ref => d_ip:8888/device/restart_daemon]
```
In most instances, it is cleaner to try to ask the server to close itself (at least try) (mysql/sub-processes are joined cleanly this way).

So, I have implemented a "close" action POST for both device and node:

* node: `/node-action/close`
* device: `/controls/<id>/close`

Systemd will restart daemons within 5s after they close.

ggilestro, 2015-09-23:
still not clear to me why you don't use pacman for updates

qgeissmann, 2015-09-23:
First of all, most of the work here is actually quite independent from whether we use pacman or git.

For instance:

* detecting devices and their status
* report when devices have been updated
* update devices ensuring it was actually stopped
* restart daemons after update
* ensure we don't break the update system whist updating (hence independent daemon)

In terms of amount of code, the github specific part of the code fits only in a few line (https://github.com/gilestrolab/ethoscope/blob/fix_update/scripts/ethoscope_updater/updater.py).
In terms of configuration files, this is virtually nothing as we just `git clone` from the right bare.

Then, let us assume developers work on github to make new features/fixes. 
We need to:

* Get the current developmental snapshot (=>latest commit on github) on the node (using internet).
* Propagate the same snapshot on all devices (using internal wireless connections).

Git makes it very straightforward as we simply set a bare repo on the node.
Then we can just fetch on bare to mirror the upstream repo. Then we can then locally use this repo as origin to propagate updates to all devices. This has several advantages:

* This is really what git is good at doing, so it is extremely simple to implement (not a lot of config to do, and a few lines of code).
* This is anonymous, so we don't need to play with ssh keys (we had issues in the past)
* We don't need to reinstall the package after update as we installed it with `pip -e` in a first place
* Very importantly, we can **work on branches**. This is fantastic when developing new features. If we do not do that, we will either **break a working platform**, or need to always have two platforms, one for development, one for stable, with different devices.
* All the versions are on the local bare repo, so we can always revert to any version/commit in case of severe bug.

We started thinking about pacman in a first place, but it does not provide all these features. Also, it requires more work and probably will introduce more bugs.

For instance, if we used pacman, we would need:

* make a decent pacman package (save static files at the right place,...). This is possible, but takes time.
* Setup a custom repo on our own server. This is possible, but time consuming. We also need to guaranty the server stays up.
* Ensure that when **any**  developer makes changes, new version gets published within minutes on the upstream pacman repo.
* work on mirroring, the remote pacman repo on the node, from the web (this is a simple git fetch atm).
* work on using custom repo on devices (a simple git pull).

Using pacman would need alteration both in the code and in the installation scripts, at it would require custom config files (we have little control on the latter as not part of the package).

In the end, I think `git` is great for the job. 
We get a immediate connection between git (which we will use to develop anyway) and our platform instead of adding another complication, probably more bugs and work. 
I think this is also why things like the `julia` package managers are, in fact, just front-ends to github.

qgeissmann, 2015-10-07:
Testing  @pepelisu 's updater:

[image]

feed back:

- [ ] Update seem to work ok on selected devices, which is really good, but **daemon not restarted yet**.
- [ ] Instead of having a **first panel to describe status of the platform** and a **second one for selection  actions**, we have a single panel. I find that confusing. The second panel should include the three buttons, and have a header showing that this is what you want to perform on the selection.
- [ ] For me, **the first panel should display available version/date** on bare/github (see schematic above).
- [ ] I don't think we should display ALL the available branches in the first panel. Just say which is the active branch. 
- [ ] nothing happens when trying to **switch branch**. This should show us available branches.
- [ ] There should be a **column up-to-date vs outdated** just after Name (sorry, not in the schematic).
- [ ] In the data table the **row for node should have Name=Node**
- [ ] when no internet connection (i.e. cannot fetch from github), no warning is shown.

closed 2015-10-28 by qgeissmann