gilestrolab/ethoscope #35: Router has limit of 15 wireless clients

qgeissmann, 2015-10-02:
Today, we started, for the first time, 15+ simultaneous tracking devices.
The last devices plugged could not connect to the wifi network.
After investigation, the only likely explanation is that our router **has a limit of 15 clients**.
Indeed, we cannot connect a 16th client. After we stop connection from one of the 15 connected clients, we can connect another device...

Also see:
http://forum.tp-link.com/showthread.php?3251-Wireless-client-connect-limit-to-15

If this is confirmed. 
It would be very sensible to acquire a new router.
Any suggestion on the model @pepelisu ?

ggilestro, 2015-10-02:
Are you using 2.4 or 5ghz? You get twice as many clients with 2.4ghz. Dual
band netgear routers should accommodate 64 devices at 2.4ghz

pepelisu, 2015-10-04:
What we can try first (or discover) is if we use a wifi access point we can reach more devices. We can use a second router (identically to the one that you have) setted up as "repeater" and see if we can get more devices. That should give us another 14 devices. 
I hoped to reach this limit much later and with more devices, but this is a portable router and the I can see that the limit is to low due to power consumption and range, with so many devices range will rapidly drop and power consumption will be reach the limit. 
A more professional router can provide more simultaneous connections but they are really expensive.
Possible something like this firewall/router: http://www.sonicwall.com/us/en/products/TZsoho.html

qgeissmann, 2015-10-13:
fixed by using more powerful router. Network still slightly troublesome though (see #38).

closed 2015-10-13 by qgeissmann