gilestrolab/ethoscope #75: Devices become "not in use"

qgeissmann, 2017-02-13:
This is only a scanning/ node specific issue.
This happens for device where IP address is `192.169.123.N`, where `N>64`.
It seems to be related to default option in the `scapy` package. should be fixed soon

qgeissmann, 2017-02-13:
So at the moment, we have more than 60 devices plugged in.
One of them has ip `192.169.123.67`. I can ping it and ssh to it -- all fine.
However, I cannot detect it using ARP tables...

I tried this:

```python
from scapy.all import srp, Ether, ARP, sr1, ICMP, IP, TCP
e = Ether(dst="ff:ff:ff:ff:ff:ff") / ARP(pdst="192.169.123.0/24")
ans, unans = srp(e, timeout=3, iface="eno1")
collection = [rcv.sprintf(r"%ARP.psrc%") for snd, rcv in ans]
print  sorted([int(c.split('.')[3]) for c in collection])

packet = IP(dst="192.169.123.67", ttl=20)/ICMP()
print sr1(packet, iface='eno1', timeout=2)
```
And get that:

```
Begin emission:
..*...*..........*....Finished to send 256 packets.
***..***********.***.*...**...*****.****......*..*...*.....*.......................*.......................................................................................................***......*.........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
Received 1028 packets, got 41 answers, remaining 215 packets
[6, 7, 8, 9, 10, 12, 13, 15, 16, 17, 18, 19, 21, 22, 24, 25, 26, 27, 28, 30, 31, 32, 35, 36, 37, 38, 39, 42, 45, 46, 49, 50, 53, 54, 56, 60, 61, 62, 63, 64, 254]
Begin emission:
.........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................WARNING: Mac address to reach destination not found. Using broadcast.
WARNING:scapy.runtime:Mac address to reach destination not found. Using broadcast.
Finished to send 1 packets.
................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
Received 1369 packets, got 0 answers, remaining 1 packets
None
```
I do get a response for another device (64):

```python
packet = IP(dst="192.169.123.64", ttl=20)/ICMP()
```

I am concerned by the warning `WARNING: Mac address to reach destination not found. Using broadcast.`
For now, I will fall back on **not using ARP table** for filtering candidate devices until we fix it.

pepelisu, 2019-06-14:
Was this a problem in scapy package or in our code? Is still there in the new version 2.4.5?

ggilestro, 2020-06-26:
This was fixed when we abandoned scapy and active scanning in favour of zeroconf / avahi - See: https://github.com/gilestrolab/ethoscope/commit/af1d394db0d2a090c9df9642d886b71f8827b030

closed 2020-06-26 by ggilestro