gilestrolab/ethoscope #78: Ethoscope stop unexpectedly

estebanbeck, 2017-05-26:
Hi, it is the second time this happen in 2 days, in two differente ethoscopes (21 and 16). Both time after less than a day of recordings.

The experiments were started on the 2017-05-25 (ethoscope 16) and 2017-05-24 (ethoscope 21)

I found the ethoscpe in the "stopped" status and the messege below was given.

Sorry there has been an error: Traceback (most recent call last): File "/opt/ethoscope-git/src/ethoscope/web_utils/control_thread.py", line 416, in run hardware_connection, StimulatorClass, stimulator_kwargs) File "/opt/ethoscope-git/src/ethoscope/utils/io.py", line 465, in __exit__ self._write_async_command(v) File "/opt/ethoscope-git/src/ethoscope/utils/io.py", line 494, in _write_async_command raise Exception("Async database writer has stopped unexpectedly") Exception: Async database writer has stopped unexpectedly 

Any thoughts?

estebanbeck, 2017-05-30:
This happened again, two times. Again the ethoscope 16 stop and the otherone was 12.

pepelisu, 2017-05-30:
A solution will be restart the service after the check if it is alive. Instead of raising an exception and stop. 
So line 492:
`    def _write_async_command(self, command, args=None):
        if not self._async_writer.is_alive():
            raise Exception("Async database writer has stopped unexpectedly")
self._queue.put((command, args))`
Should be changed to: 
`    def _write_async_command(self, command, args=None):
        if not self._async_writer.is_alive():
            self._async_write.start()
self._queue.put((command, args))`
Quentin, should check if this is actually a solution or if it will break the whole system.

qgeissmann, 2017-05-30:
thanks for reporting that.
I will have a look/ fix it when I come back from germany!

pepelisu, 2017-07-06:
This happen to me today. Is it solved in dev? or still we do not know the origin of the problem?

pepelisu, 2017-07-08:
I think it is related to rsync task or cron task to synchronize the backup with other computer. It happened to me after I made a rsync. can it be?

qgeissmann, 2017-07-10:
It is not solved and hard to reproduce.
There are two layers in the backup process:

1. `ethoscope_backup.service` should run in the node and sync MySQL -> local sqlite file. this does not involve rsync/cron.
2. The node send the data to a cloud computer (this is custom in your lab). In our case, using cronjob, twice a day (see [script](https://github.com/gilestrolab/labscripts/blob/master/node/push_ethoscope_results.sh)). This should interput and resume `ethoscope_backup.service`.

Do you think the second layer could actually break ethoscopes (e.g. by stopping the first layer and creating a "lock bug" in MySQL)?

ggilestro, 2019-09-18:
This error seems to be easily reproducible by removing binlog from mariadb (done in an attempt to reduce I/O on the SD card). A solution could, therefore, be to limit the size of the binlog adding the following to `/etc/my.cnf.d/server.cnf`

```
[server]
log-bin=mysql-bin
binlog_format=mixed
expire_logs_days = 10
max_binlog_size  = 100M
```

We'll do some testing with these new settings.

closed 2020-06-26 by ggilestro