gilestrolab/ethoscope #80: Corrupted files when using rsync to sync db with external backup.

pepelisu, 2017-07-10:
I am not sure 100% this is a bug. When the system is running if you perform a rsync with the node to have an external backup of the data, sometimes you get a broken copy of the database. So if you run rethomics to have an idea of what is going on, it stops with an error: rsqlite_fetch "Database disk image is malformed"
Do you have the same problem? 
For me it is not useful to have a backup of the database that is corrupted, could it be a rsync problem,though.

qgeissmann, 2017-07-10:
do you use: https://github.com/gilestrolab/labscripts/blob/master/node/push_ethoscope_results.sh?
It looks like you are modifying the file whilst fetching it. Ideally, you want to stop the backup daemon, rsync then restart it...

pepelisu, 2017-07-10:
I did not stop the backup daemon, so that was the problem.

qgeissmann, 2017-07-10:
we still nee to have a robust system I think, it should be the responsibility of `ethoscope_backup.service` to guaranty file integrity...

qgeissmann, 2017-11-28:
closed via #82

closed 2017-11-28 by qgeissmann