gilestrolab/ethoscope #97: Fly misdetection around cotton area

antortjim, 2019-07-23:
## Description

We have recently registered impossible fly movements, where the fly would cross huge distances in only one frame, sometime the cotton or food area, and then returns to the original place. We called this artifact “spike” (figure C ROI 18).  It is critical when happening very frequently because the 5-min-rule breaks and flat sleep traces are produced (figure A left)  Even though it’s easy to discriminate from real movement because of its magnitude (figure E left, see dots at xy_log10distx1000 > -1000), we still haven’t found the source of this issue as we have no video recordings where this issue occurs, only the resulting db file.

However, we know:

* these spikes occur almost always toward the outer side of the tube i.e the fly seems to go to coordinate 0 (left) on ROIs 1-10 and to 1 (right) on ROIs 11-20. See figure E for an example of a frame where the fly is misdetected on the cotton on RO 16. 
* they seem to happen more frequently on problematic ROIs like 1 and 15. We believe ROI 1 is problematic because it is highly sensitive misalignments of the hardcoded ROI grid to the arena (see figure D), due to for example camera inclination. Central ROIs like 14, 15 and 16 are also problematic because they have less light during the night, due to the IR strip holder lying right beneath them.

## Proposed solution at hardware level

So far, we have always arranged our experiments with black cotton, and we introduced the tubes so the food faces the interior of the arena, and the cotton the outside. This means that the ethoscope is detecting flies in the same area as the cotton. For this reason, we believe the solution is:

* Improving the lighting conditions (taking care not to produce to much heat). We propose using a 6 LED strip.
* Making sure the camera is not inclined and the ROIs are detected correctly
* Using white cotton and not introducing it too deep, so it nevers overlaps with the arena i.e. the cotton should always be in the region of the tube that sticks out of the arena.
* Eventually flip the tube, so cotton lies inside, which is what I can see from the manual, is what you do in your lab.

## Proposed solution at software level

In order to rescue these and any new data that might exhibit this issue I implemented a simple median filter (Mode_filter), upon which if the X coordinate of the fly is the same on 6 or more frames (out of a window of 10 frames), then X is set in the remaining ones to that coordinate too. It seems to work because the spikes last only 1 frame and the next coordinate is exactly the same as before the spike. It could be done more restrictive (requiring 9 frames to be the same, and not just 6 for example), but I think it’s already quite restrictive, so only artifacts and not true movements are removed.
Once x is corrected, the xy_dist_log10x1000 is set to a low enough value if the x differences are 0 so the motion_detector function in Rethomics does not flag the data as movement.

pepelisu, 2019-07-24:
Hi Antonio, 
Many thanks for this exhaustive report. 
From the hardware side I can tell you that another light_box has been develop with the aim to improve the light conditions. We use now a custom made guide panel with the LED on one side. First tests show a promising improvement. You can find the onshape file here: https://cad.onshape.com/documents/98512e7af7f4bb54b04f14ad/w/45b95c21c6563124c1461db2/e/a742f8bda55848f2cf1a9ce2
The hole thing, once printed and the acrylic guide panel added should look like this:
[image]
[image]
You can use 3 or 6 led strip, the design can be improved to use a dissipator for the leds or a longer space between flies and leds. 
If you tests it please let us know what do you think.

antortjim, 2019-07-25:
Dear @pepelisu 

Thank you for your prompt reply and your insights regarding the lightbox! I think it's a great improvement. For now, we will not print light boxes with the new version. Instead, we will replicate your design by placing 2 strips of 3 LEDs each on the short sides of the light box. We will use no dedicated diffuser other than the arena itself.

We connect the 2 strips in series, so it looks like this, using the old light box
[image]

I took a picture of how the illumination looks like on one ethoscope in the D phase with **A)** the old design (6 LEDs on the led holder) vs **B** the new design (the picture above).

**A**
[image]
**B**
[image]

The improvement is clear on the 4 central ROIs (5,6,15,16). However it seems a bit darker.
I am running a sleep deprivation experiment with this new setup now. I will let you know if the results!

PS I think on the other hand, more sophisticated image processing methods could be implemented, so the system is more robust to the lighting conditions? As long as the RPi can take it... or maybe we can try the Jetson Nano (<100 Euros) in future versions?

Thank you once again!!
Antonio

antortjim, 2019-09-25:
Dear Gilestro lab

I have a follow up on tracking issues.
We replaced the LED strips on all our ethoscopes as shown above. The amount of issues has diminished, but we still encounter tracking issues.
We observe a strange behavior by which the tracker places flies closer and closer to the left end of the tube (closer to 0). The resulting shape of the x trace is a 'right triangle' shape.
We firmly believe this is a mistracking as a fly is highly unlikely to display such a strange moving pattern.
Below you can see the x coordinate trace for all flies, and a zoom-in between hours 41 and 42 for flies 1 and 16. For fly 16 for example, I would say the fly did not move at all from around 0.62, but the tracker registers this increasingly bigger spikes towards 0, trumping the motion_detector in R.

PS
This issue is the reason why I posted issue #104 

[image]
[image]
[image]

ggilestro, 2019-09-30:
I agree that this shape looks artifactual. It also appears to be synced in time across flies in the same tube. To me, this may suggest some external light disturbance coming from the incubator such as visible LED flickering in the incubator or someone opening the door. Have you considered these trivial possibilities first? At any rate, my recommendation is the same as the other issue you opened: record a 24 or 48 hours long video then do offline tracking of the same. Please let me know.

antortjim, 2019-10-02:
I ran an ethoscope with modifications in the code to save the intermediate steps of the processing and the result. As you mentioned in #104, doing this greatly slows down tracking from ~10FPS to ~0.3 FPS. I had already tried the video but I didn't get any result.
Fortunately, in this case, I was able to replicate the error:
[image]

and I was able to playback the processing steps of the tracker, as seen in the gif below.
The spacer in the middle is placed there to improve the contact between the tubes and the shafts of the motors in the optomotor.
[image]
Its frames and many more can be found here:
https://mega.nz/#F!8hFywaTR!IRVA8cwAZU1L4iLQokSDMQ

Indeed, the flickering of the IR light is very significant and I think it's driving the tracking errors. I can see that the flickering is translated into the edges of the ROIs is sometimes included in the foreground mask (black&white figure). More concretely, spurious pixels above the threshold of the mask (20/255) seem to lie around the edge of the spacer. The edge of the fly food in the tube also has this effect but less pronounced I would say.
This seems to happen only when the fly has not moved for a while. Since the fly is quiet, it produces no changes in the foreground and thus smaller changes in the background caused by light flickering become big enough to confuse the tracker. But if the fly moves, even if the light flickers, the signal produced by the fly offsets this noise and the tracker succeeds.

Have you ever faced such an issue before? Did you redesign the lightbox for a similar reason?

A list of things I think can be done to attempt to solve this.

**Hardware**
1. Make lighting more robust. Maybe the IR strips are not reliable, or maybe the RPi cannot provide a stable energy supply. In that case, a more robust PSU would be the solution.

**Software**
1. Increase the foreground masking threshold, so it becomes more difficult for spurious pixels to be flagged as foreground.
https://github.com/gilestrolab/ethoscope/blob/f692e7c973b54e1aa1580f198aa836fb7faa99b6/src/ethoscope/trackers/adaptive_bg_tracker.py#L383

2. Define a threshold for feasible velocities between frames (taking into account the time between them), so huge velocities i.e. 'teleportations across the tube' due to mistracking is filtered out.

3. Make sure the x,y coordinates that are saved are actually the center of mass of the ellipse and not the center of mass of the foreground-background (I can see it's a todo)
https://github.com/gilestrolab/ethoscope/blob/f692e7c973b54e1aa1580f198aa836fb7faa99b6/src/ethoscope/trackers/adaptive_bg_tracker.py#L468
This is achievable if the ellipse is drawn on black canvas (i.e. `np.zeros_like(self._buff_fg)` and not the `self._buff_fg` object, which contains the spurious pixels)

3. An ideal solution at the software level would be to upgrade the tracking algorithm to make it more robust, maybe with a CNN but probably it would be too much for the RPi to take.

Thank you for your help beforehand!
Antonio

antortjim, 2019-10-02:
One sidenote! I checked that the slowdown effect is due to me producing the figures shown in the gif. If I just change the period of `ImgToMySQLHelper`  to 0.1, i.e. I save all snapshots but I don't save intermediate steps, the resulting FPS is ~3, which I think is fine (way better than 0.3).

antortjim, 2019-10-03:
I created this gif with the snapshots I saved during the debugging run. Every snapshot represents 1-3 seconds. It's clear that the overall light changes a lot between frame, but I don't know the cause yet. But at least this confirms the flickering is not an artifact of the algorithm i.e. it could be due to malfunctioning of either the camera, the IR strip or the PSU (the RPi and in turn an Anker PowerBank). 

[image]

ggilestro, 2019-10-03:
I think there are two possibilities here:

1) a hardware issue. Perhaps the LEDs are defective of your system does not provide enough current. Make sure you use a power supply that is powerful enough (at least 2A) and maybe try powering the IR LEDs not through the PI. We use 12V rated IR LEDs powered with the 5V coming from the PI

2) a software issue. The picamera library got some recent changes to solve compatibility issues with the PI4. This created some [sporadic issues with the piNOIR camera](https://github.com/raspberrypi/firmware/issues/1167). You may want to try disabling the auto white balance in the python code. If you do, you need at the same time to adjust gains otherwise the image will be black. Check the [API](https://picamera.readthedocs.io/en/release-1.10/api_camera.html) and the [recipes](https://picamera.readthedocs.io/en/release-1.12/recipes1.html). 

If you see a purple image under white light, this may be due to the software issue. I'd rather be oriented to think it is a hardware one, though.
Please let me know how this goes.

antortjim, 2020-01-20:
Dear @ggilestro 

Thanks for your response. We are now encountering new artifacts in our data:

[image]

Small fluctuations of the position of the fly are detected when using the python3.7 branch. They are very small (< 1/100 the width of the tube) but the detected corrected velocity becomes greater than 0.003 (the default coefficient in the motion_detector function), which thus causes the data to be annotated as awake, even though the data is not moving. We do not want to increase the correction coefficient for consistency with previous experiments, but I have instead tried:

```
linear_filter <- function(x, kernel) {
  result <- as.integer(filter(x, kernel))
  result[is.na(result)] <- x[is.na(result)]
  return(result) 
}

dt_raw <- load_ethoscope(
  metadata = metadata,
  result_dir = "/media/antortjim/data/antortjim/ethoscope_results/"
)

dt_raw[,xy_dist_log10x1000 := as.integer(linear_filter(x = xy_dist_log10x1000, kernel = linear_kernel)), by = 'id']

dt <- sleep_annotation(dt_raw, 0.003)

```

antortjim, 2020-01-21:
Following up on this, I just checked the 6000th datapoint of the annotated dataset for region id 1 in this dataset. You can see here that even though x and y don't change in many contiguous rows, the distance indeed changes and sometimes becomes higher than log10(0.003) * 1000, which is the default threshold of the `max_velocity_detector` function -> https://github.com/rethomics/sleepr/blob/cb4efa653100d215f1cb31ad7da4a505d266dd3d/R/motion_detectors.R#L32

**See for example the 8th and 9th rows. x and y do not change but the distance becomes greater than -2522.879 (log10(0.003) * 1000). I don't understand how is this possible since the distance only relies on x, y and the width of the ROI**
https://github.com/gilestrolab/ethoscope/blob/0567585158ff3a51ea998faa97a8d60285711163/src/ethoscope/trackers/adaptive_bg_tracker.py#L455
which stays between 465-466 pixels.

The xy_dist_log10x1000_manual column is manually created by me, following the formula of the Euclidean distance and assuming x and y are already normalized with w_im (as x and y oscillate between 0 and 1 on this table).

```
w_im <- 465
points_list <- as.list(dt_raw_point[, .(x, y)])
points_list <- lapply(1:length(points_list[["x"]]), function(i) c(points_list$x[[i]], points_list$y[[i]]))

xy_dist_log10x1000_manual <- c()
for (i in 2:length(points_list)) {
  difference_between_points <- (points_list[[i]] - points_list[[i-1]])
  # print(difference_between_points)
  res <- sqrt(sum(difference_between_points**2))
  res <- round(log10(1/w_im + res) * 1000)
  xy_dist_log10x1000_manual[i] <- res
}
```

[image]

[image]

As I understand `max_velocity_detector`, it:
  - takes this xy_dist_log10x1000
  - transforms it back into a distance,
  - computes the time difference between frames
  - divides distance with the time difference to get the velocity
  - corrects the velocity, which means multiplying again by the time difference (thus getting the distance again), and then divides by the velocity correction coefficient (0.003)
  - selects the maximum over a block of a given time (by default 10 seconds)
  - if the maximum velocity is greater than 1 i.e. the max distance > 0.003, the block of 10 seconds is annotated as moving.

Thank you very much for your time.

qgeissmann, 2020-01-21:
Hi there, thanks for reporting your issue!
> See for example the 8th and 9th rows. x and y do not change but the distance becomes greater than -2522.879 (log10(0.003) * 1000). I don't understand how is this possible since the distance only relies on x, y and the width of the ROI

Keep in mind that, internally (before the saving to the db), x and y are subpixel accurate as they are computed from the centre of mass of the greyscale detected object (`y,x = ndimage.measurements.center_of_mass(self._buff_fg_backup)`). However, to save space in the database, `y` and `x` are saved as integers (`x_var = XPosVariable(int(round(x)))`).
Therefore, in the DB, `xy_dist_log10x1000` can change even when `x` and `y` don't.  

You said the problem is in the python3 branch... I wonder if this could be due the default behaviour of a division of ints in python2 vs 3 (`/` vs `//`)?

antortjim, 2020-01-22:
Hi @qgeissmann thanks for your reply

Indeed, I did not think of the difference between `/ `and `//` in python2 and python3. However, a glance at the line that computes the distance and in general the function it is in tells me that should not be a problem

https://github.com/gilestrolab/ethoscope/blob/0567585158ff3a51ea998faa97a8d60285711163/src/ethoscope/trackers/adaptive_bg_tracker.py#L470

as the code enforces the division between float types. I don't see any other place where a conflict between `//` and `/` could arise.

Could you list the specifications of the ethoscopes you are using with your python3.7 branch? That would help us a lot!
We are using:
* python 3.7.4
* scipy 1.3.1
* opencv 4.1.1
* numpy 1.17.2
* RaspberryPi 3B+
* Pi camera V2.1
* add anything you may think could be a culprit

this is the output of `uname -a` for our ETHOSCOPE_001
```
Linux ETHOSCOPE001 4.19.75-2-ARCH #1 SMP PREEMPT Tue Oct 1 00:34:07 UTC 2019 armv7l GNU/Linux
```
Our code is that in your python3.7 branch with 2 modifications:
I uncommented this line 
https://gitlab.com/antortjim/ethoscope/blob/fsl_python3/src/scripts/device_server.py#L447 because the ethoscope 'kept waiting for a network'

And I added a method to  the ControlThread class to handle an error with the camera that causes it to be 'out of resources' i.e. taken by a previous process.

Both changes are listed on this commit:
https://gitlab.com/antortjim/ethoscope/commit/c0c95fad4b9dc1ccea3e34a3b58a7507e3b66c1c

This is the picamera error. It eventually causes the frame grabber to timeout, which is the error we get in the GUI.
```
Jan 21 13:40:50 ETHOSCOPE013 python3[611]: picamera.exc.PiCameraMMALError: Failed to enable connection: Out of resources
Jan 21 13:40:50 ETHOSCOPE013 python3[611]:     raise PiCameraMMALError(status, prefix)
Jan 21 13:40:50 ETHOSCOPE013 python3[611]:   File "/usr/lib/python3.7/site-packages/picamera/exc.py", line 184, in mmal_check
Jan 21 13:40:50 ETHOSCOPE013 python3[611]:     prefix="Failed to enable connection")
Jan 21 13:40:50 ETHOSCOPE013 python3[611]:   File "/usr/lib/python3.7/site-packages/picamera/mmalobj.py", line 2212, in enable
Jan 21 13:40:50 ETHOSCOPE013 python3[611]:     self.renderer.inputs[0].connect(source).enable()
Jan 21 13:40:50 ETHOSCOPE013 python3[611]:   File "/usr/lib/python3.7/site-packages/picamera/renderers.py", line 558, in __init__
Jan 21 13:40:50 ETHOSCOPE013 python3[611]:     self, self._camera.outputs[self.CAMERA_PREVIEW_PORT])
Jan 21 13:40:50 ETHOSCOPE013 python3[611]:   File "/usr/lib/python3.7/site-packages/picamera/camera.py", line 513, in _init_preview
Jan 21 13:40:50 ETHOSCOPE013 python3[611]:     self._init_preview()
Jan 21 13:40:50 ETHOSCOPE013 python3[611]:   File "/usr/lib/python3.7/site-packages/picamera/camera.py", line 433, in __init__
Jan 21 13:40:50 ETHOSCOPE013 python3[611]:     capture = PiCamera()
Jan 21 13:40:50 ETHOSCOPE013 python3[611]:   File "/opt/ethoscope-device/src/ethoscope/hardware/input/cameras.py", line 353, in run
Jan 21 13:40:50 ETHOSCOPE013 python3[611]:     self.run()
Jan 21 13:40:50 ETHOSCOPE013 python3[611]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
Jan 21 13:40:50 ETHOSCOPE013 python3[611]: Traceback (most recent call last):
```

Again thanks for your help!

ggilestro, 2020-02-27:
Hi Antonio

Do I understand correctly that you are using a version that was forked from GitHub to GitLab in November? I made several hundred commits in the past 4 months so it is going to be difficult for me to reproduce this or future problems unless we converge again!

Having said this: I do not remember having touched anything in the tracking algorithm when migrating from python 2 to python 3. Given you see one-pixel fluctuations I suppose this may be either a consequence of rounding a value that is close to 0.5 (stochastically floored or ceiled) or again due to some lighting artifact (the fact you see this only at night makes me opt for the latter hypothesis).

If it is an issue with rounding, it may be related to the [intrinsic limitations of floating-point arithmetic.](https://docs.python.org/dev/tutorial/floatingpoint.html#tut-fp-issues)

We could fix that by fixing the rounding procedure to one direction.

How frequent is the artifact?

antortjim, 2020-03-02:
Hi @ggilestro 
Thank you for looking into it :)

Yes, I am using a forked version, but I merge your changes on my fork when I see a significant milestone is achieved. We don't update very frequently because we want to be conservative. The last commit I have from you is

```
commit 0567585158ff3a51ea998faa97a8d60285711163
Author: ggilestro <giorgio@gilest.ro>
Date:   Fri Nov 22 17:58:41 2019 +0000

    Added back an internal function that was removed by mistake. This handles the special occurrence of the img table
```
Indeed, we also thought these two factors could be contributing to the problem. However, at the turn of the new year I updated our ethoscopes and also those used in the Verstreken la, which were fabricated independently. And then both labs started experiencing the problem very frequently, to the point of 

preventing us from doing more experiments. The traces indicated that the flies were barely sleeping at all in the night. The fact that all their ethoscopes were exhibiting the same problem at the same time, made me think a buggy IR strip is unlikely to be the reason.

A look at the detected `max_velocity`es showed that these were somehow systematically getting shifted upwards, just enough to overcome the velocity threshold of 0.003. You can see that more than half of the background ‘non-moving’ max_velocities are > 1 in the dark phase, indicating that the non corrected velocity was > 0.003. This did not happen with ethoscopes running the python2 version, and it explains the observed traces that indicate lack of sleep in the night.

[image]

We came up with an easy solution, which was adding a high pass filter (only let through > 730 nm) to the RPi camera, so the lighting conditions in L and D were identical in terms of what the camera ‘sees’.
We cropped this filter https://www.robertwhite.co.uk/lee-filters-100mm-system-polyester-87-infrared-filter.html
into tiny pieces and mounted it on top of the camera, like this

[image]

This has the effect of making the background velocities similar again in L and D i.e. no more upwards shift.
We combined this with a change in the software in Python and R, so the ethoscope saves x and y with float precision in the SQL database and R uses this offline to recompute the distance, as indicated here
R: https://github.com/rethomics/sleepr/issues/7#issuecomment-588143906
Python:

* Float variables definition https://gitlab.com/antortjim/ethoscope/-/blob/fsl_python3_precise/src/ethoscope/core/variables.py#L136
* Usage in the adaptive_bg_tracker class: https://gitlab.com/antortjim/ethoscope/-/blob/fsl_python3_precise/src/ethoscope/trackers/adaptive_bg_tracker.py#L492 

It is working fine for us like this now, even though the analysis exactly as we did before i.e. using the distance stored in the sqlite3 file and 0.003 as threshold still shows this problem. Using the sqlite3 distance and 0.006 as threshold rescues the data in some cases, I don’t know what fraction. I explain that by the fact that the distance in R is always gonna be 1/roi_width less than in Python (I have another workaround for the log(0) error), so it is more easily scored as non movement.
We will run calibration experiments to verify that these modified settings are correct, similar to what you did before you found 0.003 was the best threshold. For us it seems it is not, and we haven’t found the reason yet.
Hope I was clear enough!

Best regards,
Antonio

ggilestro, 2020-03-03:
Hi @antortjim 

thank you again for reporting bugs and changes in such a careful manner. This is extremely useful. After looking into this, I think this is actually a bug that affects ethoscopes that are using the newest camera version (`pinoir v2`) on python3.

This probably all originated from this external issue: https://github.com/raspberrypi/firmware/issues/1167

One thing we can try, is to turn autowhitebalance off on these lines

https://github.com/gilestrolab/ethoscope/blob/9ae58e8fdabeb694c1d4968baa921c534a3e140f/src/ethoscope/hardware/input/cameras.py#L356-L362

Would you like to give it a try and let me know?

antortjim, 2020-03-05:
This is how our frames look on an ethoscope in the L phase _without the IR filter I posted above_ if I don't convert to gray

[image]

and the D phase
[image]

placing the IR filter makes the L phase image look like in D. The image is purple in the D phase, but I don't think that's hardware malfunction, just the nature of IR light, right?

I tried setting
```
capture.awb_mode = 'off' 
```
and indeed the output is a black square. I am not setting custom gains, as explained in the links you provided, due to lack of time (for now).

On a RPi 3B+ with camera Noir v2.1 running
python-3.7.4-2
picamera 1.13

I got the above pictures running the script below and accessing `http://192.169.123.61:9000/static/tmp/last_img.png` (that's the IP of ETHOSCOPE_011). Streaming by clicking on the button available in the Python3 GUI I get a similar picture. 
```
import picamera
from picamera.array import PiRGBArray
import logging
import cv2

target_fps=20
target_resolution=(1280, 960)

with picamera.PiCamera() as capture:

    logging.warning(capture)

    capture.resolution = target_resolution
    capture.awb_mode = 'auto'
    #capture.awb_mode = 'off'

    capture.framerate = target_fps
    raw_capture = PiRGBArray(capture, size = target_resolution)
    for frame in capture.capture_continuous(raw_capture, format="bgr", use_video_port=True):

        try:
            raw_capture.truncate(0)
            #out = cv2.cvtColor(frame.array, cv2.COLOR_BGR2GRAY)
            out = frame.array
            cv2.imwrite('/tmp/last_img.png', out)

        except Exception as e:
            logging.warning(e)
            break
```

antortjim, 2021-02-07:
Just remembered I did not report on our solution to the problems here:

* We put the IR filter on the camera so data from L and D phase have similar properties.
* We power our IR strip with an external PSU to avoid flickering effects caused by unstable current provided by the RPi. I don't know what causes this unstability, but an external PSU solves the issue. It means though the ethoscope now needs two cables: one to power the RPi and one to power the strip.
* We use a different velocity correction coef. Apparently the switch to Python 3 caused some of the algorithm that processes the frames and yields arrays in Python changed. The Python code is the same but the output is not.
@ggilestro mentioned adding this line
```
awb_auto_is_greyworld=1
```
to `/boot/config.txt` revers the algorithm back to the same as in Python2. That is the same solution they found here https://github.com/raspberrypi/firmware/issues/1167
That should also work

closed 2021-02-07 by antortjim