gilestrolab/ethoscope #98: Better documentation for the distance computation

antortjim, 2019-07-23:
Dear Gilestro lab

I would like to better understand how you compute the distance traveled by flies between 2 consecutive frames. After analyzing your code, I can see:

* first, you treat the x,y pair as a complex number.
https://github.com/gilestrolab/ethoscope/blob/f692e7c973b54e1aa1580f198aa836fb7faa99b6/src/ethoscope/trackers/adaptive_bg_tracker.py#L470
I think this in practice implies loking at changes in x only, ignoring changes in y?

* also, in the `_track` method of the `AdaptiveBGModel` tracker:
https://github.com/gilestrolab/ethoscope/blob/f692e7c973b54e1aa1580f198aa836fb7faa99b6/src/ethoscope/trackers/adaptive_bg_tracker.py#L473

 This is in charge of taking the old x,y coordinates and the new ones and computing a relative distance. Nevertheless, I don't understand it. I would expect something like an Euclidean distance, but I don't think that is the case.

I see you substract the pair of coordinates (which are relative to the window width i.e x=0.5 means halfway through the tube), then take the absolute value. I understand the log10 x 1000 operation serves to get a number that is easy to work with if I am not wrong. But I don't understand why do you add 1/roi_width to the absolute difference of relative coordinates?! Using this width again would only make sense to me if it was to multiply the relative coordinates to get again an absolute distance i.e. number of pixels traveled.

I looked in the suppl. material of the Plos One paper as well Mr. Geissmann PhD thesis, but I couldn't find an explanation on this. Could you give one here?
Thank you very much in advance and once again thanks for making the ethoscopes free and libre :)

Best regards,
Antonio

qgeissmann, 2019-07-23:
Hi @antortjim, 

Thanks for your question! Regarding the first part of your question, both x and y positions are used. The use of complex numbers is simply an arrithmetic trick to compute the euclidean distance and store position in a single variable:

```
sqrt((x_1 - x_0)^2 +(y_1 - y_0)^2) == abs(x_1 +1.0j*y_1 - x_0 +1.0j*y_0)
```

In other words, the absolute difference (ie, the modulus) between two complex numbers is the euclidean distance between the two points: `abs(pos - self._old_pos)`.

For instance, try:

```
pos_0 = 1 +1.0j*1 # x=1, y=1
pos_1 = 2 +1.0j*2 # x=2, y=2
abs(pos_0 - pos_1) # == 1.4142135623730951 # == sqrt(2)
```
Then, as you have noticed, the position is immediatly divided by the width of the image, so x and y are between 0 and 1:  `pos /= w_im`. 

Now, there is a rather high risk that, numerically, the position has not changed at all. Therefore, instead of calculating the log10 of the relative displacement, which could result in a numerical error (`log10(0)`), we add one pixel, which in the relative terms is `1/w_im`. 

I hope it makes more sense, sorry for the confusion,
Let me know,
Quentin

antortjim, 2019-07-24:
Dear @qgeissmann 
Thank you for your fast answer and the good explanation.
I see that when `abs` is passed a complex number, it computes the modulus of the vector that the number represents on the real-complex plane, thus it will do a computation that is after all identical to Pythagoras' theorem.
I can only add a parenthesis to your line of code ;)
`sqrt((x_1 - x_0)^2 +(y_1 - y_0)^2) == abs(x_1 +1.0j*y_1 - (x_0 +1.0j*y_0))`

Also, I now understand why you added this relative pixel unit.
Thanks again!
Antonio

closed 2019-07-24 by qgeissmann