gilestrolab/ethoscope #99: Ethoscope identity: editing the /etc/machine-id file

antortjim, 2019-07-25:
Dear gilestrolab

**TL;DR**

I don't know how to customize the string contained in /etc/machine-id while at the same time keeping journalctl working. Changing it manually breaks journalctl in our ethoscopes. Do you know how to avoid breaking journalctl? How do you do it? Thank you!

**Long version**

I would like to change the content of the /etc/machine-id file to a custom string that reflects the ethoscope's identity, similar to what you did on your ethoscope public dataset here https://zenodo.org/record/1068324

However, I encountered issues trying to open the file with write access, even as root with either nano (as shown in the manual https://qgeissmann.gitbooks.io/ethoscope-manual/building-and-installation/ethoscope-from-scratch.html#card-identity) or vi.
My solution was to overwrite its contents by `echo`ing `$CUSTOM_STRING > /etc/machine-id`.
However, I have noticed that doing so breaks the systemlog i.e. after doing so journalctl shows no entries, even after reboot.
```
[root@e05 alarm]# journalctl
No journal files were found.
-- No entries --
```

A run of `[root@e05 alarm]# systemctl status systemd-journald*`shows this:

```
* systemd-journald.socket - Journal Socket
   Loaded: loaded (/usr/lib/systemd/system/systemd-journald.socket; static; vendor preset: disabled)
   Active: failed (Result: service-start-limit-hit) since Wed 2016-07-20 13:43:12 UTC; 3 years 0 months ago
     Docs: man:systemd-journald.service(8)
           man:journald.conf(5)
   Listen: /run/systemd/journal/stdout (Stream)
           /run/systemd/journal/socket (Datagram)

* systemd-journald.service - Journal Service
   Loaded: loaded (/usr/lib/systemd/system/systemd-journald.service; static; vendor preset: disabled)
   Active: failed (Result: start-limit-hit) since Wed 2016-07-20 13:43:12 UTC; 3 years 0 months ago
     Docs: man:systemd-journald.service(8)
           man:journald.conf(5)
  Process: 206 ExecStart=/usr/lib/systemd/systemd-journald (code=exited, status=1/FAILURE)
 Main PID: 206 (code=exited, status=1/FAILURE)

* systemd-journald-dev-log.socket - Journal Socket (/dev/log)
   Loaded: loaded (/usr/lib/systemd/system/systemd-journald-dev-log.socket; static; vendor preset: disabled)
   Active: failed (Result: service-start-limit-hit) since Wed 2016-07-20 13:43:12 UTC; 3 years 0 months ago
     Docs: man:systemd-journald.service(8)
           man:journald.conf(5)
   Listen: /run/systemd/journal/dev-log (Datagram)

```
The only way I found to fix journalctl is by running

```
rm /etc/machine-id
systemd-machine-id-setup 
```
as indicated by this answer https://serverfault.com/a/800047
but this creates a random string, and I can't figure out the way to create the one I want, _while keeping journalctl working_ .

PS Here https://qgeissmann.gitbooks.io/ethoscope-manual/building-and-installation/ethoscope-software.html#burning-the-sd-card---the-easy-way you refer to a link to download your ethoscope images, but I can't find any link there. The only image I found is one of the files in https://zenodo.org/record/831154#.XTow03UzY5k. Is a link in   https://qgeissmann.gitbooks.io/ethoscope-manual/building-and-installation/ethoscope-software.html#burning-the-sd-card---the-easy-way missing?

ggilestro, 2019-07-26:
You can manually edit the `/etc/machine-id` file as long as you fill it with a properly formatted UUID. See the [UUID format description on Wikipedia](https://en.wikipedia.org/wiki/Universally_unique_identifier#Format). 

Any wrongly formatted UUID will break some systems (such as journalctl) and will be replaced by a random, properly formatted one at boot.

I have now restored the links to the new images. Please be aware that those refer to the Python3.7 version which is still experimental at this stage. We are now beta testing it in the lab and will become master probably sometimes in September. With that new version, we assigned ethoscopes a custom UUID which is a three digits number representing the ethoscope XXX + the remaining digits of a randomly generated UUID using the following code:

`new_uuid = "%03d" % id + uuid4().hex[3:]`

where `id` is the ethoscope number

antortjim, 2019-07-26:
Thank you for your answer!
The way we do it is by having a fixed suffix string, generated by running the UUID generator and replacing 2 characters with the initials of our PI. There's another lab here that also uses the ethoscopes and they have their PI's initials on their ethoscopes machine ids. We devised this system so we could easily keep track of whose was each ethoscope.
The suffix looks like this:
`e27SL5a9e19f94de287e28f789825`
to which we preprend a three digit number encoding the ethoscope number (similar to you)
`001` for example

This probably has a wrong format, as journalctl stops working with it. But the string is not reset at boot, it stays the same. However, I have noticed that if I remove one of the characters from there and reboot, indeed the string is replaced by a random one and journalctl works again. 

I just tried replacing the suffix by the output of `uuid4().hex[3:]` and then it works. So my conclusion is that we are somehow breaking the format by replacing the output of the random generator. Replacing SL to sl does not fix it either.
I am now trying to find a suffix that allows us to distinguish labs and does not break the format. 

Thank you once again!
Antonio

antortjim, 2019-07-26:
I can see for example that 
`0053d5fc56c647888c88180666ce5833` 
is valid, whereas
`0053d5sl56c647888c88180666ce5833`
i.e. replacing fc for sl, renders it invalid!
I will try some other combinations and see how can I get it to work.
Thank you

pepelisu, 2019-07-26:
UUID will take only hexadecimal characters: ABCDEFG012345689
FC is within this characters, SL is not.
You can change any character but only for one of those.

ggilestro, 2019-07-26:
@antortjim  - you may want to consider using the machine-name instead of the machine-id. That is a human-readable custom file that is read only by the ethoscope platform and you'll have the possibility to change it in any way you want.

closed 2019-07-26 by ggilestro

antortjim, 2019-07-26:
I read several times it was restricted to hexadecimal characters, but I didn't realise what it meant >.< Many thanks for taking the time to answer a bit of a dumb question!
@ggilestro we like customising the `/etc/machine-id` because that's the name of the toplevel folder of each ethoscope under `/ethoscope_results`. I will discuss with the other lab what we should do. For now, I can say I tried replacing the SL with AA and I can confirm it works!
Thank you all!
Antonio