gilestrolab/ethoscope #103: Wifi connectivity on fresh install

matt-p-brown, 2019-09-24:
Hey Gilestro Lab,

Thank you all for making this great system available. I'm excited to get it up and running. I've set up our first ethoscope, router, and node according to the instructions on your manual. I am having a problem with my ethoscope connecting to the network set up on the router though both wifi (top picture) and ethernet (bottom). To help troubleshoot, I hooked the pi up to a monitor and took pictures of the steps where it seems to have problems on startup. I'll apologize in advance, I have little to no experience with archlinux or python. Thanks for the consideration.

Best,
Matt

Wifi alone
[image]

With eithernet hookup to pi
[image]

ggilestro, 2019-09-24:
Hi @matt-p-brown - there was an issue with an earlier version of the SD image linked in the manual. This has been fixed last week. Please burn again your SD card using the most recent version ([20190918](https://imperialcollegelondon.box.com/s/gugx1sn3nmf3p4zlmxn766uuvrtnqjml) as I write) and that should solve this problem.

closed 2019-09-24 by ggilestro

matt-p-brown, 2019-09-25:
Thanks so much for the prompt response!