gilestrolab/ethoscope #108: Problems formatting the SD card

DenaFrancis, 2020-03-03:
Hi,
I have been trying to format the SD card so that I can begin building the ethoscope. I am a biologist, so not much coding experience. 

I am trying to format my SD card on a Mac (macOS Mojave 10.14.6)

I am using a card reader hooked up to the computer, and the SD card adapter is on the unlocked position. My SD card is a Samsung Evo 32GB.

This is what the terminal looks like, its a modified archlinux script that I am following from your manual:
disk2 is where the SD card is on my computer. 
Deannes-MBP:~ dena$ SD_CARD=/dev/disk2
Deannes-MBP:~ dena$ echo "o p n p 1 +100M t c n p 2 w"| fdisk $SD_CARD
fdisk: /dev/disk2: Permission denied
I also tried this to have a look:

ls -l /dev/disk2
brw-r----- 1 root operator 1, 9 1 Mar 16:14 /dev/disk2s1
Not sure if it helps.

I have looked at other posts/problems online, and I have tried to blow into the card reader (someone said that it helped). But it didn't do anything.

When I look at sharing and permissions for the SD card, it says you have custom access.
I have asked others online to help with this issue, but the answers were complicated and did not result in a successful outcome. 

I am not sure what to do since I invested some of my own money for the parts and would like to proceed with building this. 
Thanks

qgeissmann, 2020-03-03:
Hi, not much of a mac user, but most likely you need admin (supersuser) rights to format your card (https://support.apple.com/en-us/HT202035). Try opening a superuser session `$ sudo su`. **BE VERY CAREFUL** then. Especially ensure `/dev/disk2` is indeed your sd card and not another drive (that you will break).

DenaFrancis, 2020-03-03:
Hi Quintin,
Yes, I made sure that disk2 is the right one by looking at diskutil list and it had the right size and name. I tried that and this is what I got: 

SD_CARD=/dev/disk2
Deannes-MBP:~ dena$ echo "o

> p
> n
> p
> 1
> +100M
> t
> c
> n
> p
> 2
> w"|sudo fdisk $SD_CARD
Password:
Disk: /dev/disk2	geometry: 3891/255/63 [62521344 sectors]
Signature: 0xAA55
         Starting       Ending
 #: id  cyl  hd sec -  cyl  hd sec [     start -       size]
------------------------------------------------------------------------
 1: 0C 1023 254  63 - 1023 254  63 [      2048 -     204800] Win95 FAT32L
 2: 83 1023 254  63 - 1023 254  63 [    206848 -   62314496] Linux files*
 3: 00    0   0   0 -    0   0   0 [         0 -          0] unused      
 4: 00    0   0   0 -    0   0   0 [         0 -          0] unused 

Does this look right to you and should I proceed with the rest of the code? If I can. I am not sure that terminal on Mac has the ability to execute some of the mkfs.vfat code. 
Thanks for your help,
Dena

ggilestro, 2020-03-03:
Please do not do this. Use instructions provided on the ethoscope manual online: https://qgeissmann.gitbooks.io/ethoscope-manual/building-and-installation/ethoscope-software.html

You should burn the SD card using balena-etcher and it will be ready to go.

closed 2020-03-03 by ggilestro

DenaFrancis, 2020-03-03:
Hi Dr Gilestro,

I find that balena-etcher doesn’t support .tar.gz files. I got this message when I tried to drop the ethos cope image file 

/Users/dena/Downloads/ethoscope_core_files (1)/ethoscope_os_20160721.tar.gz is not a supported image type

Sorry about that, seems that it's more complicated for me? Or I am not doing something correctly. I went onto their GitHub site and asked about this issue. But they haven’t responded yet.  

Thanks,
Dena

ggilestro, 2020-03-03:
You are not supposed to burn that file. Just follow the instructions on the online manual, verbatim.

[image]

The SD image you need to download is linked in the manual ([it's this one](http://lab.gilest.ro:8001/latest_sd_image))

DenaFrancis, 2020-03-03:
Okay, thank you. I was using the older version of the manual then.