gilestrolab/ethoscope #110: Dead link http://lab.gilest.ro:8001/latest_sd_image

BanoRo, 2020-03-10:
Dear gilestrolab,

The link http://lab.gilest.ro:8001/latest_sd_image for the installation of the ethoscope seems to be dead. Could you please fix this?

Thank you!
Best,
BanoRo

ggilestro, 2020-03-11:
Done.

ggilestro, 2020-10-22:
This is a problem with port 8001, I think. It must be firewalled by some universities and occasionally some users will not be able to connect to the resources page.

pgorelik, 2021-01-04:
The link appears to be dead to me too. I have tried this on multiple networks, including at my university and at home. I have also tried playing with firewall settings, and I am unable to get through. I would appreciate some help to get the latest RPi image.

Thanks!
Pavel

ggilestro, 2021-01-04:
That link points to a redirect. The current version (as of 4/1/2020) is [here](https://imperialcollegelondon.box.com/shared/static/l2aq5cj1wug9u8hd8xhdw8t12xrs39u1.zip).

That link actually works for me but it seems to be firewalled, especially in the USA. I should probably just change the port from 8001 to something less exotic.

ggilestro, 2024-01-31:
This is now fixed with commit 4768bc0b436c10c0d54800dfa406e47205e4191b
The new URL for the resource page is https://ethoscope-resources.lab.gilest.ro

closed 2024-01-31 by ggilestro