gilestrolab/ethoscope #112: Node crashes after latest update.

DenaFrancis, 2020-04-30:
Hi, I am using the raspberry pi 4b to run the node after burning an image file onto an sd card. I have run into a problem after updating the node to the latest (3/12/2020 version) on the ethoscope updater( http://node:8888). After rebooting the node, it looks like the node is no longer working and cannot be accessed online. 
[image]

The administration tab is blank:
[image]

I can, however, still ssh into the node and it seems to be connected to the ethoscope and the internet as I can ping these and there is a signal.

nigelb, 2020-04-30:
My Node also fell over after upgrading.

I now get the following errors in the system log:

```
Apr 30 05:56:45 node python[25596]: Traceback (most recent call last):
Apr 30 05:56:45 node python[25596]:   File "/opt/ethoscope-node/node_src/scripts/server.py", line 16, in <module>
Apr 30 05:56:45 node python[25596]:     from ethoscope_node.utils.device_scanner import EthoscopeScanner, SensorScanner
Apr 30 05:56:45 node python[25596]: ImportError: cannot import name 'EthoscopeScanner' from 'ethoscope_node.utils.device_scanner' (/usr/lib/python3.7/site-packages/ethoscope_node-trunk-py3.7.egg/ethoscope_node/utils/device_scanner.py)
Apr 30 05:56:45 node python[25595]: Traceback (most recent call last):
Apr 30 05:56:45 node python[25595]:   File "/opt/ethoscope-node/node_src/scripts/backup_tool.py", line 2, in <module>
Apr 30 05:56:45 node python[25595]:     from ethoscope_node.utils.backups_helpers import GenericBackupWrapper, BackupClass, receive_devices
Apr 30 05:56:45 node python[25595]: ImportError: cannot import name 'receive_devices' from 'ethoscope_node.utils.backups_helpers' (/usr/lib/python3.7/site-packages/ethoscope_node-trunk-py3.7.egg/ethoscope_node/utils/backups_helpers.py)
Apr 30 05:56:45 node python[25594]: ERROR:root:Traceback (most recent call last):
Apr 30 05:56:45 node python[25594]:   File "/opt/ethoscope-node/node_src/scripts/video_backup_tool.py", line 117, in <module>
Apr 30 05:56:45 node python[25594]:     server )
Apr 30 05:56:45 node python[25594]: TypeError: __init__() takes 4 positional arguments but 5 were given
Apr 30 05:56:45 node audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=ethoscope_node comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
Apr 30 05:56:45 node systemd[1]: ethoscope_node.service: Main process exited, code=exited, status=1/FAILURE
Apr 30 05:56:45 node systemd[1]: ethoscope_node.service: Failed with result 'exit-code'.
Apr 30 05:56:45 node systemd[1]: ethoscope_backup.service: Main process exited, code=exited, status=1/FAILURE
Apr 30 05:56:45 node systemd[1]: ethoscope_backup.service: Failed with result 'exit-code'.

```

ggilestro, 2020-04-30:
This looks like an issue with the updater not properly building the new Python Egg.
Using the terminal. please go inside the `ethoscope/node_src` folder and issue the following command:

`sudo python setup.py develop`

nigelb, 2020-04-30:
Yep that does seem to have fixed it.

DenaFrancis, 2020-04-30:
Yes, it now shows that everything (including the ethoscope is updated). Would we need to do this after every installation? or will the newer versions fix this? Thanks a lot.

DenaFrancis, 2020-05-01:
Just an update: after the node and ethoscope software was updated, the ahavi-daemon in the ethoscope was not able to run after rebooting the system. I went back to clear the journal logs again and the ethoscope reconnected after that.  I can't seem to save/add any user/experiment information through the node though...and I am not sure how to fix this.

commit 76f4a179 (2020-05-01): After pulling the GIT update generates a local python egg using the `python setup.py develop` command.
Fixes #112
closed 2020-06-06 by DenaFrancis