gilestrolab/ethoscope #113: Video Recording Error

KandimallaPrat, 2020-05-05:
Hi, 
I just set up the node and a couple of ethoscopes. The live tracking works as expected - saving the txt and db files. However, I am unable to record videos for offline tracking. 
Attached is the screenshot of the issue I have been having. While the recording seems to start (I can see a live stream on the page), there is nothing listed in the Run ID - the video is not saved anywhere.

[image]

Any suggestions on how to fix this?

pepelisu, 2020-05-06:
I could not find it now in the documentation, but in order to be able to save the videos, the service `ethoscope_video_backup.service` has to be started in the node: `systemctl start ethoscope_video_backup.service`
Videos will be saved to the default path (if nothing else specified): `/ethoscope_data/videos`
In order to help you further, please paste the status of the ethoscope_video_backup service.

ggilestro, 2020-05-06:
Hi @KandimallaPrat - the video is working and it is saved in chunks of h264 files in a local folder in the ethoscope. The path looks something like this: 
```
[root@ETHOSCOPE144 2020-05-06_09-53-32]# pwd 
/ethoscope_data/results/144683573e2a4c1588e1b23553a24eca/ETHOSCOPE_144/2020-05-06_09-53-32
[root@ETHOSCOPE144 2020-05-06_09-53-32]# ls -lh
total 25M
-rw-r--r-- 1 root root  18M May  6 09:58 2020-05-06_09-53-32_144683573e2a4c1588e1b23553a24eca__1280x960@25_00000.h264
-rw-r--r-- 1 root root 6.1M May  6 10:00 2020-05-06_09-53-32_144683573e2a4c1588e1b23553a24eca__1280x960@25_00001.h264
```
To transfer those videos to the node you would have to make sure your ethoscope_video_backup.service is running

You can check that's the case in the node administration page:

[image]

If the backup service is running, you would see the video appear on your node, as @pepelisu  just wrote.

I agree that the output of the ethoscope window could be a bit more clear regarding the state of this. I have just made some changes to the UI to make it more explicit and I'll commit it soon.

closed 2020-05-06 by ggilestro

KandimallaPrat, 2020-05-08:
Hi @pepelisu and @ggilestro, 
Thank you very much for the prompt clarification. 
1) The ethoscope_video_backup service has been active during the recording sessions. 
[image]

2) I have just updated the node. 
[image]
I also checked to make sure that the ethoscope_video_backup service is active in the Node Administration page. 

I still run into a similar issue. 
[image]
The video is not saved anywhere on the Node (the node administration page shows the path as /ethoscope_data/videos . 

I then checked the local folders on the ethoscope: 
total 0
This is the case for each recording session that I tried over the past 2 days. 

Again, thank you very much for your help!

ggilestro, 2020-05-08:
That's weird. I'll need more info to debug this.
Can you post a copy of the logs for both the node and the ethoscope?

KandimallaPrat, 2020-05-08:
Please find attached the complete logs for the Ethoscope and the Node.
[Node Log - PK.pdf](https://github.com/gilestrolab/ethoscope/files/4601160/Node.Log.-.PK.pdf)
[Ethoscope Log - PK.pdf](https://github.com/gilestrolab/ethoscope/files/4601161/Ethoscope.Log.-.PK.pdf)

pepelisu, 2020-05-08:
Is it possible that the ethoscope memory (SD card) is already full of video content? 
There is this line that is worrying me.
`May 08 10:27:59 ETHOSCOPE001 python3[943]: OSError: [Errno 28] No space left on device`

ggilestro, 2020-05-08:
Yes - @pepelisu is right. You are using a version of the SD image in which the ethoscope_data folder is mapped to the root partition instead of the data partition and therefore it fills up with video very quickly. I have actually fixed that earlier in May so if you burn a new image you should be alright. Sorry about that.

KandimallaPrat, 2020-05-08:
Hi @pepelisu and @ggilestro,
Thank you very much! The ethoscope log shows that the root partition is 98%full. I just assumed that everything was being saved in the mmcblk0p3 partition which is 23GB and only 2% used. 
I will burn the new image. 
Thank you very much for your help.