gilestrolab/ethoscope #115: Cannot add user data to node interface

DenaFrancis, 2020-06-06:
Hi gilestro lab,
I am running the node from a RPI4 and I have updated the latest version of the node and ethoscope (from May). But I am having an issue with the management page of the node interface. It won't let me save the user information or save the path of where to put the data. However, I can save new incubator information. After typing in the user info, the save button is greyed out.
[image]

Also, I managed to run a pilot tracking experiment. I let it track for a few days, but I don't know where to find that data. I have a pic of the experiments, but nothing that goes for three days. 
[image]

Thanks!

ggilestro, 2020-06-07:
Regarding the first issue (users). 
The first three fields are actually required even though PIN and email are not currently used. They will be used at some point to send notifications to users if there is a problem with the platform.
I have now changed the page to have some more explicit information about that.

[image]

Regarding the second aspect: make sure your ethoscope-backup daemon is actually running.
Click on the gear under "management" and the first line should tell if your ethoscope backup is running or not.

DenaFrancis, 2020-06-07:
Thank you for the fast response! 
For issue one: I have added the information into the fields, but it still won't save?
[image]

For issue 2: I think that the backup daemon is active, at least in terms of what is displayed.

[image]
Thank you

ggilestro, 2020-06-08:
For issue 1: try your full name - a username will be created for you (Dena Francis -> dfrancis)
For issue 2: I don't quite remember now how I configured the node on the rPi4. If the backup is active, you should be having data in the `/ethoscope_data/results` folder.

DenaFrancis, 2020-06-08:
Issue 1: resolved. Thank you! 
Issue 2: I checked the ethosocope results files:

```
[root@ETHOSCOPE001 results]# du -hs /ethoscope_data/results/*
646M /ethoscope_data/results/00155fa83cc848019df1c9029bfb899f
14K	/ethoscope_data/results/001916f944d54aaa9b7743793dc0bd16
8.0K	/ethoscope_data/results/index.html
```
And on the node:
`[root@node node]# du -hs /ethoscope_data/*
8.0K	/ethoscope_data/results
4.0K	/ethoscope_data/tmp
601M	/ethoscope_data/videos
[root@node node]# du -hs /ethoscope_data/results/*
4.0K	/ethoscope_data/results/results_200527_003815.zip`

This file was created on a previous test, so it doesn't have the 3-day test results.

ggilestro, 2020-06-10:
Hi Dena - 

Are you using an external hard drive to save the data? The rPI SD card is not large enough to be used for data storage. You should follow the manual and mount an external hard drive instead.

closed 2020-06-13 by DenaFrancis