gilestrolab/ethoscope #116: CPU temp 0°C

posttenebre, 2020-06-20:
**Issue**
During tracking experiments, the CPU temperature reading is 0°C

**Ethoscopes**
Raspberry pi 3 B+ 
[python3.7], 853fee... (2020-06-07 18:57:05)

**Main node**
Raspberry pi 4 B
[python3.7], 853fee... (2020-06-07 18:57:05)

**Description**
This is not a major issue but it would be nice if I can fix why there is no value for CPU temperature in the tracking page. I can't find any reference to 'temperature', 'CPU', 'vcgencmd', or 'measure_temp' in any of the source code.

ggilestro, 2020-06-20:
mmmh. Sorry, I can't reproduce this.
CPU temperature is read by this function:

https://github.com/gilestrolab/ethoscope/blob/853fee65da5b0895b436902dd91fd1874ef4aa4f/src/ethoscope/web_utils/helpers.py#L313-L325

If you get a zero it could be one of the following options:
1. machine is not detected as a PI for some reason (different chipset?!)
2. vcgencmd is not present at that location
3. some other issue I can't think of

posttenebre, 2020-06-20:
I suspect that it's due to the Pi not being detected. In 'Info and Logs' I get
`PI version : 0`
`PI camera : false`

Output of `cat /proc/cpuinfo` shows:
`Hardware	: BCM2835`
`Revision	: a020d3`
`Serial		: 000000005d377ac6`
`Model		: Raspberry Pi 3 Model B Plus Rev 1.3`

Would adding the following lines to ethoscope/src/ethoscope/web_utils/helpers.py solve the issue?
`    elif hardware.group(1) == 'BCM2835' and 'a020d3' in revision.group(1):`
`       # Pi 3`
`      return 3`

posttenebre, 2020-06-20:
Ok so quickly changing the ethoscope/src/ethoscope/web_utils/helpers.py does change how the raspberry pi is discovered. 
`PI version : 3`
`PI camera : Run tracking once to detect the camera module`

and 

`CPU temperature: 46.2 ℃`

Thanks for the suggestions. Still not sure why I can't search in github for the words Temperature and actually find that file.

posttenebre, 2020-06-20:
#117

ggilestro, 2020-06-20:
Hi @posttenebre  - thanks a lot for this. Yes, your proposed solution would work but I decided to rewrite that function from scratch to be more future proof. It turns out there currently are ~30 PI revisions and the new code should cover all of them, be future proof, and provide a more descriptive output.

closed 2020-06-20 by ggilestro

posttenebre, 2020-06-20:
No worries. Thanks for the help.