gilestrolab/ethoscope #120: No access to ethoscope settings fields

jfriaspt, 2020-07-02:
Do not have access to ethoscope settings (options - > settings). Not able to configure ethoscope id.

[image]
[image]
[image]

The node is running on Virtualbox machine and detects ethoscope000. Before it worked, but after updates not anymore. I can configure ethoscope machine-name and hostname local files by ssh access and bypass this step, but after that not able to start tracking with similar issue (no options fields).  

[image]
[image]

I have no clue why this is happening. I'm a biologist and this is not my expertise. 
Btw the ethoscope project is really essential to the science community and I want to congratulate all developers envolved in this.

ggilestro, 2020-07-02:
Thanks for pointing this out. This was a small bug on the ethoscope side. If you update your ethoscope, it will go away.

closed 2020-07-02 by ggilestro

ggilestro, 2020-07-02:
Fixed also in [today's SD image](https://imperialcollegelondon.box.com/shared/static/ckkhkaoor93xxwr4a0jb3c79htrhn7jy.zip).

jfriaspt, 2020-07-02:
Thanks very much for the support! It works fine now .