gilestrolab/ethoscope #122: Optomotor Module stops with the following error

budhaChowdhury, 2020-07-14:
[image]

The Optomotor Module starts up and runs without any problems, but after overnight tracking, exits with the error message.

qgeissmann, 2020-07-14:
Hi @budhaChowdhury, thanks,

Does it happen systematically? Does it happen when just tracking? It looks like this is an exception when the input image (within a region of interest) is completely black (mean = 0)...
if it is transient (e.g. the camera output a few black frames to start with), we could skip the errors by changing the code to something like:

```py
try:
    scale = 128. / mean[0]
except ZeroDivisionError:
      raise NoPositionError
```

budhaChowdhury, 2020-07-14:
Hi Quentin, 
Thanks for your feedback! 
The problem is arising when the lights in the incubator turn back on (respective day) after tracking in the dark (respective night). I just ran the module a couple more times and every single time the tracking stops as the lights are turned back on, even with a lower intensity. Do you think I should still try out the altered code?

ggilestro, 2020-07-14:
Are you positive the IR lights are working? this is the kind of error you
would expect if IR lights were not functioning but in that case you would
expect the error to pop up shortly after lights off. If things are as you
suggest, on the other hand, I suppose what is happening is that visible
lights create a strong flash of light and the camera closes the shutter as
a response, resulting in some pitch black frames.

I don't think we've ever seen that error. We can work around it using the
solution Quentin suggested but if you have an IR lights issue you'd still
lose data.

qgeissmann, 2020-07-14:
Yes, I would check that the lights are working as well (maybe just take a video). But I would say the *untested* patch I propose should make the tracking more robust in the case this is just a spurious couple of frames... That would perhaps be smart for us to implement so that long experiments stop after several days of acquisition... That said, it is still very hard to generate completely black frames, which is why we never had this issue...

budhaChowdhury, 2020-07-14:
Yep the IR lights are working - I am attaching a screenshot. 
I ran a couple more times again - it quits with the same error message as the lights turn back on... Quentin can you walk me through how to use the patch? Thanks! 
ps. Should I also try a different camera? 
[image]

qgeissmann, 2020-07-14:
@ggilestro Is the update service still running on the new version? I would suggest adding the patch on a github branch so that @budhaChowdhury can update devices to this specific branch for testing...

ggilestro, 2020-07-14:
Yes the update service runs as before but just commit this to `dev` : it's
safe enough. Thanks Quentin. Good to see you on board :)

qgeissmann, 2020-07-14:
I don't have much infrastructure to test anything, but I am happy to help anything back-endish :). I can make a PR on dev and you can test.

qgeissmann, 2020-07-14:
once the patch goes through to dev, you can update following [this information](https://www.notion.so/Updating-ethoscopes-0771914363c9408b988c3d62a3972b22). Ensure you pick the `dev` branch, not `master`. That would be very helpful if you let us know how it goes :),

budhaChowdhury, 2020-07-14:
Thank you so much, will definitely keep you updated how it goes!

budhaChowdhury, 2020-07-15:
Hi Quentin, 
in the dev branch (for ethoscope) the status says  "Software broken"- should I go ahead and update it anyway? Thanks!
ps. I am guessing I should update the Ethoscope, and not the node?

ggilestro, 2020-07-15:
I am traveling and I have not accepted the modification request yet. Please
wait 24h.

budhaChowdhury, 2020-07-15:
okay, just checking : ) Thanks a ton!

closed 2025-06-27 by ggilestro