gilestrolab/ethoscope #125: Main node won't update via interface.

posttenebre, 2020-07-16:
**Issue**
When trying to update the main node from the :8888 web interface I get a `connection refused` error.

**Ethoscopes**
Raspberry pi 3 B+
[python3.7] a9047f... (2020-07-02 08:02:50)

**Main Node**
Raspberry pi 4 B
[python3.7] a9047f... (2020-07-02 08:02:50)

**Other**
Unlike the described set-up in the manual, I have chosen to use the PoE hats with a PoE switch without a WIFI router.

**Error Message**
`Traceback (most recent call last): File "/usr/lib/python3.8/urllib/request.py", line 1350, in do_open h.request(req.get_method(), req.selector, req.data, headers, File "/usr/lib/python3.8/http/client.py", line 1240, in request self._send_request(method, url, body, headers, encode_chunked) File "/usr/lib/python3.8/http/client.py", line 1286, in _send_request self.endheaders(body, encode_chunked=encode_chunked) File "/usr/lib/python3.8/http/client.py", line 1235, in endheaders self._send_output(message_body, encode_chunked=encode_chunked) File "/usr/lib/python3.8/http/client.py", line 1006, in _send_output self.send(msg) File "/usr/lib/python3.8/http/client.py", line 946, in send self.connect() File "/usr/lib/python3.8/http/client.py", line 917, in connect self.sock = self._create_connection( File "/usr/lib/python3.8/socket.py", line 808, in create_connection raise err File "/usr/lib/python3.8/socket.py", line 796, in create_connection sock.connect(sa) ConnectionRefusedError: [Errno 111] Connection refused During handling of the above exception, another exception occurred: Traceback (most recent call last): File "/opt/ethoscope_updater/update_server.py", line 178, in group response = updates_api_wrapper(device['ip'], device['id'], what='device/update') File "/opt/ethoscope-node/scripts/ethoscope_updater/helpers.py", line 300, in updates_api_wrapper raise e File "/opt/ethoscope-node/scripts/ethoscope_updater/helpers.py", line 286, in updates_api_wrapper f = urllib.request.urlopen(req) File "/usr/lib/python3.8/urllib/request.py", line 222, in urlopen return opener.open(url, data, timeout) File "/usr/lib/python3.8/urllib/request.py", line 525, in open response = self._open(req, data) File "/usr/lib/python3.8/urllib/request.py", line 542, in _open result = self._call_chain(self.handle_open, protocol, protocol + File "/usr/lib/python3.8/urllib/request.py", line 502, in _call_chain result = func(*args) File "/usr/lib/python3.8/urllib/request.py", line 1379, in http_open return self.do_open(http.client.HTTPConnection, req) File "/usr/lib/python3.8/urllib/request.py", line 1353, in do_open raise URLError(err) urllib.error.URLError: <urlopen error [Errno 111] Connection refused>`

**Description**
When trying to update the main-node I receive a `connection refused` error message. This does not occur when updating the ethoscopes. The manual solution is to perform a git pull on the main-node.

`sudo git pull origin python3.7`

This would suggest that the node is capable of performing the update but that the script performing the update is getting snagged on something. Does the updater perform a specific check for an internet connection that might result in errors due to my non-standard set-up?

ggilestro, 2020-07-20:
Thanks for reporting this. I am abroad now and not able to debug this until late August but I'll try to have a look as soon as I get my hands on a PI4. @pepelisu - have you seen this before?

posttenebre, 2020-07-20:
Thanks for the response @ggilestro. This is not high priority for me as I can still update via SSH. Let me know if you have any ideas of things I test.

pepelisu, 2020-07-21:
Hi, I did not see this before, but I can see that it is failing when trying to open the url for updates. In this file scripts/ethoscope_updater/helpers.py#L269 there is a call to this url `"http://{ip}:{port}/{what}/{id}".format(ip=hn, port=port, what=what, id=id)` 
 I suggest to construct this manually and then try to connect to it. What I think is that the ip taken for the node is wrong or the port 8888 is closed. Maybe the RPI4 is taking the global ip address and then it is not possible to reach port 8888, this should be possible with the local ip address.

posttenebre, 2021-01-26:
@pepelisu I'm back at this and I apologise for not answering sooner. I think the issue is that the main node does not have an ID. 

[image]

posttenebre, 2021-01-27:
I have also tried `http://192.168.7.100:8888/check_update/node` but no luck.

ggilestro, 2021-02-18:
I will need to have a look at this on a pi4.  The id being marked as `N/A` is not a problem. Nodes have no id.

roaldarbol, 2021-11-30:
I have a similar issue: When I try to update, nothing happens - my error message is different though. 

**Main Node**
Raspberry pi 4 B
`20201126_ethoscope_000.img`

**Other**
Currently also a slightly different setup. Connecting the node to internet with an ethernet cable changed the IP address to *that* network instead of `ETHOSCOPE_WIFI`; now I just use my internet router and ssh over that network that instead of on `ETHOSCOPE_WIFI`. It also might be worth mentioning that I still haven't mounted an external harddisk.

---

```bash
[alarm@node ~]$ systemctl status ethoscope_update_node
* ethoscope_update_node.service - Ethoscope node server
     Loaded: loaded (/usr/lib/systemd/system/ethoscope_update_node.service; enabled; vendor preset: disabled)
     Active: active (running) since Sat 2020-10-03 18:13:47 UTC; 1 years 1 months ago
   Main PID: 303 (python)
      Tasks: 13 (limit: 4915)
     CGroup: /system.slice/ethoscope_update_node.service
             |-303 /usr/bin/python /opt/ethoscope_updater/update_server.py -g /opt/ethoscope-node -b /srv/git/ethoscope.git
             |-373 git cat-file --batch-check
             `-374 git cat-file --batch

Nov 30 09:46:47 node python[303]:  = [up to date]      2flies-haar        -> origin/2flies-haar
Nov 30 09:46:47 node python[303]:  = [up to date]      dev.obsolete       -> origin/dev.obsolete
Nov 30 09:46:47 node python[303]:  = [up to date]      gh-pages           -> origin/gh-pages
Nov 30 09:46:47 node python[303]:  = [up to date]      gray_rois_mask     -> origin/gray_rois_mask
Nov 30 09:46:47 node python[303]:  = [up to date]      holes_tracking     -> origin/holes_tracking
Nov 30 09:46:47 node python[303]:  = [up to date]      master             -> origin/master
Nov 30 09:46:47 node python[303]:  = [up to date]      python3.7          -> origin/python3.7
Nov 30 09:46:47 node python[303]:  = [up to date]      qgeissmann-patch-1 -> origin/qgeissmann-patch-1
Nov 30 09:46:47 node python[303]:  = [up to date]      roi_from_mask      -> origin/roi_from_mask
Nov 30 09:46:47 node python[303]: error: object directory /tmp/yaourt-tmp-node/aur-ethoscope-node/ethoscope-node/objects does not exist; check .git/objects/info/alternates
```

The same error also shows up in the log
```
Nov 30 10:52:38 node python[531]: error: object directory /tmp/yaourt-tmp-node/aur-ethoscope-node/ethoscope-node/objects does not exist; check .git/objects/info/alternates
```

I also attempted the above solution `sudo git pull origin python3.7` but get: 
```bash
[alarm@node ~]$ sudo git pull origin python3.7
fatal: not a git repository (or any of the parent directories): .git
```

And lastly I tried @pepelisu's idea of using http://192.168.0.24:8888/check_update/node, but:
[image]

ggilestro, 2021-12-02:
please use the `dev` branch. Other branches are too much behind (I should merge).

roaldarbol, 2021-12-03:
Will give that a try. Do you have any idea about the error when trying to update from the UI?
```bash 
Nov 30 10:52:38 node python[531]: error: object directory /tmp/yaourt-tmp-node/aur-ethoscope-node/ethoscope-node/objects does not exist; check .git/objects/info/alternates
```

ggilestro, 2021-12-03:
Ok, I found what the problem is. The node package was not installed properly in the rpi4 image and the git information did not propagate to their final destination. You can fix with the following command from within the node. 

```
su
cd /opt/
rm -rf ethoscope-node
git clone https://github.com/gilestrolab/ethoscope.git
mv ethoscope ethoscope-node

cd /opt/ethoscope-node
git checkout dev
git remote set-url origin /srv/git/ethoscope.git
cd /opt/ethoscope-node/node_src
python setup.py develop
```

I just made a new image which I will be uploading soon.

roaldarbol, 2021-12-03:
Ah, brilliant! Thanks a ton, also for being so quick in replying @ggilestro - I really appreciate it! I'll try it sometime next week.

ggilestro, 2021-12-03:
New image available [here](https://imperialcollegelondon.box.com/s/wxgmlkk2llj6xp1ww50n16c9pdid4cy8)