gilestrolab/ethoscope #126: New error when start tracking

SofiPol, 2020-08-04:
Hi! I'm having this error when I start with the tracking, some idea about what can be the problem?

Sorry there has been an error: Traceback (most recent call last): File "/opt/ethoscope-device/src/ethoscope/web_utils/control_thread.py", line 452, in run self._start_tracking(cam, result_writer, rois, TrackerClass, tracker_kwargs, hardware_connection, StimulatorClass, stimulator_kwargs) File "/opt/ethoscope-device/src/ethoscope/web_utils/control_thread.py", line 318, in _start_tracking self._monit.run(result_writer, self._drawer) File "/opt/ethoscope-device/src/ethoscope/core/monitor.py", line 134, in run raise e File "/opt/ethoscope-device/src/ethoscope/core/monitor.py", line 123, in run result_writer.write(t, track_u.roi, data_rows) File "/opt/ethoscope-device/src/ethoscope/utils/io.py", line 509, in write self._initialise(r, dr) File "/opt/ethoscope-device/src/ethoscope/utils/io.py", line 581, in _initialise self._create_table(table_name, fields) File "/opt/ethoscope-device/src/ethoscope/utils/io.py", line 625, in _create_table self._write_async_command(command) File "/opt/ethoscope-device/src/ethoscope/utils/io.py", line 619, in _write_async_command raise Exception("Async database writer has stopped unexpectedly") Exception: Async database writer has stopped unexpectedly

Thanks!
Cheers,

ggilestro, 2020-08-12:
```
Exception: Async database writer has stopped unexpectedly
```
This normally indicates an issue with the SD card: either the card is damaged or is not fast enough.
Please replace with another SD card (ideally different brand) and see if it fixes the issue.

SofiPol, 2020-08-12:
Hi Giorgio Thanks for your reply, now i'm having a different problem, I
already changed the SD card and flashed the last image, i;m able to make
streaming and video recording but when i start tracking the device become
offline and nothing happens, the SD card is the red one Samsung EVO+ 32GB I
don;t know what could be the problem, I restarted the node, the ethoscope
and nothing chage. Some ideas?

Thanks!
Cheers,
Sofia

ggilestro, 2020-08-12:
Try with a different power supply. I suppose the tracking activates the camera, causes a power surge and the power  supply can't give enough current. That's the only thing I can think of anyway.

SofiPol, 2020-08-12:
Ok Giorgio Thanks, I will try that, I'm using the Anker power supply with 5
USB but now I only have two ethoscopes connected to it...I will try to
change to a different power supply and see what happens,
Thanks,
Cheers,
Sofia

Georges-Farkouh, 2020-08-13:
I had a similar problem and I was able to fix it by replugging the camera cable into the raspberry

SofiPol, 2020-08-18:
Hi, I tried to replug the camera but I still having a problem, I rebooted
the ethoscope, also restarted the Node and I changed the SD card and I'm
not able to start tracking, I can make video recordings or streaming but
when I started tracking the ethoscope goes offline and then nothing
happens...The SD card that i'm using is the new one the 32G EVO+.
Could you help me with this?

Thanks,
Cheers,
Sofia

Georges-Farkouh, 2020-08-18:
Could you post an image of how the ethoscope goes offline please 
Georges

SofiPol, 2020-08-18:
20200818_082031.mp4
<https://drive.google.com/file/d/1rsvrRD9pj1hFpEov8Ydv1hGC5QBS8AdY/view?usp=drive_web>
Here is a video of what happens with the ethoscope, as you can see it
starts iinitialising and then it comes offline and that's all...

ggilestro, 2020-08-18:
@SofiPol - is this something that happens with all your ethoscopes or just that one?

SofiPol, 2020-08-18:
At this moment I only have two ethoscopes and it's happening with both

antortjim, 2020-09-09:
I think the best way to debug the problem is plugging a monitor to the HDMI interface of the RPi (next to the micro USB used by the PSU). Then plug the PSU and you will see the RPi turn on, display the booting process and eventually give you a prompt where you can login. Then try start tracking and see what happens in the monitor. Is the ethoscope still on? Or just offline? Maybe there is an error message displayed? Or maybe it just powers off? This experiment can help figuring it out.

Please note you must plug the HDMI first and only then the PSU. Otherwise, the display will be black,  unless the line
`force_hdmi_hotplug=1`
is already in the /boot/config.txt file of the SDcard.

Alternatively, try using the official RPi supply or unplugging the second ethoscope.

Hope it helps!

pepelisu, 2020-09-10:
It would also help to know which version of the RPI you are using. Just to isolate odd behavior with untested versions (Rpi4)

SofiPol, 2020-09-10:
Thanks to all of you, I will try to check that, I'm not having a monitor
now but I will find one, and the version that i'm using is the RPi3 A+ .

Thanks!
Cheers,
Sofia

pepelisu, 2020-09-11:
If the RPi are going offline while starting tracking all the time, I would think that the model A with only 512MB of RAM is not enough, and they are crashing while starting. The recommended model is RPI3 B+ with 1GB RAM.
I just checked mine and they are using 330MB of RAM while tracking (128MB is shared with the video processor and not usable) so 512MB - 330MB -128MB = 54MB. As soon as they need a little bit extra to start tracking they will freeze and appear to be offline and off. Only a restart will bring them up again. 
I would suggest to change the RPI for version 3B+

closed 2023-06-29 by ggilestro