gilestrolab/ethoscope #128: No video output from record video or start streaming options.

IllustratedMan-code, 2020-09-02:
Whenever I hit the "record video" button from the ethoscope menu of the node it shows a buffering sign, but no video. There is no file output either. When I ssh into the node, the video folder is empty. The "start streaming" button results in this image:
[image]

Other strangeness: There is video output when I use the start tracking option; However, the video only outputs in black and white (don't know if this is intentional). 
I installed the node as a package on an arch installation on a raspberry pi 4 as the image that was provided to burn didn't seem to even boot (ethoscope image was fine).

Any Idea what is going on?

GenRevo89, 2020-09-04:
I'm having the same issue. I am able to get a stream while tracking but not otherwise, making it very difficult to focus the pi's for new builds. I am also unable to get the video backup service to work when trying to just record video to run in CTRAX.

GenRevo89, 2020-09-04:
My node is installed on a virtual box arcolinux setup and on a bootable arcolinux as well. Both facing same problems.

IllustratedMan-code, 2020-09-04:
@GenRevo89 I ended up focusing mine by using an sd card with [raspberry pi os](https://www.raspberrypi.org/downloads/
) installed and the script below:

```

import io
import picamera
import logging
import socketserver
from threading import Condition
from http import server

PAGE="""\
<html>
<head>
<title>Raspberry Pi - Surveillance Camera</title>
</head>
<body>
<center><h1>Raspberry Pi - Surveillance Camera</h1></center>
<center><img src="stream.mjpg" width="640" height="480"></center>
</body>
</html>
"""

class StreamingOutput(object):
    def __init__(self):
        self.frame = None
        self.buffer = io.BytesIO()
        self.condition = Condition()

    def write(self, buf):
        if buf.startswith(b'\xff\xd8'):
            # New frame, copy the existing buffer's content and notify all
            # clients it's available
            self.buffer.truncate()
            with self.condition:
                self.frame = self.buffer.getvalue()
                self.condition.notify_all()
            self.buffer.seek(0)
        return self.buffer.write(buf)

class StreamingHandler(server.BaseHTTPRequestHandler):
    def do_GET(self):
        if self.path == '/':
            self.send_response(301)
            self.send_header('Location', '/index.html')
            self.end_headers()
        elif self.path == '/index.html':
            content = PAGE.encode('utf-8')
            self.send_response(200)
            self.send_header('Content-Type', 'text/html')
            self.send_header('Content-Length', len(content))
            self.end_headers()
            self.wfile.write(content)
        elif self.path == '/stream.mjpg':
            self.send_response(200)
            self.send_header('Age', 0)
            self.send_header('Cache-Control', 'no-cache, private')
            self.send_header('Pragma', 'no-cache')
            self.send_header('Content-Type', 'multipart/x-mixed-replace; boundary=FRAME')
            self.end_headers()
            try:
                while True:
                    with output.condition:
                        output.condition.wait()
                        frame = output.frame
                    self.wfile.write(b'--FRAME\r\n')
                    self.send_header('Content-Type', 'image/jpeg')
                    self.send_header('Content-Length', len(frame))
                    self.end_headers()
                    self.wfile.write(frame)
                    self.wfile.write(b'\r\n')
            except Exception as e:
                logging.warning(
                    'Removed streaming client %s: %s',
                    self.client_address, str(e))
        else:
            self.send_error(404)
            self.end_headers()

class StreamingServer(socketserver.ThreadingMixIn, server.HTTPServer):
    allow_reuse_address = True
    daemon_threads = True

with picamera.PiCamera(resolution='640x480', framerate=24) as camera:
    output = StreamingOutput()
    #Uncomment the next line to change your Pi's Camera rotation (in degrees)
    #camera.rotation = 90
    camera.start_recording(output, format='mjpeg')
    try:
        address = ('', 8000)
        server = StreamingServer(address, StreamingHandler)
        server.serve_forever()
    finally:
        camera.stop_recording()

```

Just ssh into the pi, start the script, then go to http://<raspberry pi ip address>:8000. There should be a high framerate stream of the camera.

ggilestro, 2020-09-04:
Thanks for reporting this. I cannot reproduce this but I have been looking into upgrading the streaming performance in any case so I'll most likely change things as a consequence.
For now, if you need streaming to fix the focus you could use @IllustratedMan-code 's suggestion or follow the instructions on the user manual ([here](https://www.notion.so/giorgiogilestro/Adjust-focus-on-the-camera-1abdf96db0c8417988e0f7e6147c0fe7))

GenRevo89, 2020-09-04:
Any solution for getting the video backup service to work?

ggilestro, 2020-09-07:
> Any solution for getting the video backup service to work?

This seems like a different problem. 
Can you please open a separate issue and describe exactly what the problem is?

Follow the guidance [here](https://www.notion.so/giorgiogilestro/Software-issues-7de6d3569a014c4eb7f14bb8d5a77d4e).

pepelisu, 2020-09-09:
It does work with an old version of chrome Version 79.0.3945.117 (Official Build) (64-bit).
However I can reproduce the error on updated firefox and chromium. 
Accessing the url for streaming seems to work in updated chromium 
http://node.domain.de/device/0089f1eddc2a498fad9bfa8f2da4785b/stream
However it throws an unhandled error on firefox. 

Can this be related to a cross reference error? on firefox I am getting this message"
```
 timestamp
Cookie “MYSAPSSO2” will be soon treated as cross-site cookie against “http://node.domain.de/device/0089f1eddc2a498fad9bfa8f2da4785b/data” because the scheme does not match. data
Cookie “MYSAPSSO2” will be soon treated as cross-site cookie against “http://node.domain.de/node/timestamp” because the scheme does not match.
```

IllustratedMan-code, 2020-09-09:
I haven't tried it on chrome, only firefox. I will retry with chrome today to see if that makes any difference.

IllustratedMan-code, 2020-09-14:
It doesn't work on chrome either

IllustratedMan-code, 2020-09-21:
On an outdated ethoscope (which I have unfortunately already updated) an image was displayed for the "start streaming" option on firefox, but there was no video.

ggilestro, 2020-10-20:
Which branch are we talking about, sorry? Is it `master`?
I cannot reproduce this on Firefox 81.0 on Linux. It works for me using the `dev` branch. This reminds me that I should probably merge if people use `master`.

IllustratedMan-code, 2020-10-26:
I believe it is on the master branch. I will try on the dev branch once my experiement ends. Thank you for all the help so far.

IllustratedMan-code, 2020-10-26:
It works on the dev branch on chromium and firefox on ubuntu 20.04 so I am going to go ahead and close the issue.

closed 2020-10-26 by IllustratedMan-code