gilestrolab/ethoscope #130: Camera is not cleanly release after streaming video

pepelisu, 2020-09-10:
If after streaming the video a tracking is starting one get the error of empty queue for devices. A restart solves the problem.

ggilestro, 2020-09-11:
The problem appears to be with this line:

https://github.com/gilestrolab/ethoscope/blob/013c1cd38a21c958cca7f4b55ed1bd1b3830cbb5/src/ethoscope/web_utils/record.py#L122-L125

`serve_forever` does not start a thread and the process never leaves that if statement, so it cannot be stopped and the camera is not released. I am working to fix it.

ggilestro, 2020-09-11:
This is now fixed. Please note that I completely changed the way we handle streaming so one needs to upddate both the ethoscope AND the node.

commit 8f0b738c (2020-09-11): Changes the way we handle streaming. Now the ethoscope acts as a socket server on port 8887 and waits for connection
from the node, acting as a client. We no longer need to create another http server instance and this makes stream faster and more reliable.

This also fixes #130
closed 2020-09-11 by ggilestro