gilestrolab/ethoscope #133: tolerances motor cover & motor holder - new SD module

rjs360, 2020-10-07:
Just checking before I make some slight modifications to the STLs: is anyone else having tolerance issues with 3d printed motor covers not fitting on the motor holder part (unless you want to hammer them in place without the possibility of removing them easily)?  I am referring to the sleep deprivation module parts that are described here: https://www.notion.so/New-modular-structure-and-new-electronics-a478fcbff07c469db1b531a7f50d336d 

I am printing on a not-so-high-end Creality3D Ender 3, but never had any issues with tolerances of the previous installments of the ethoscope parts.  Thanks for any feedback.

ggilestro, 2020-10-08:
In what orientation are you printing the cover?
It should be printed in this way:

[image]

rjs360, 2020-10-08:
Yes, that is exactly the orientation in which I print them.  I tried another filament yesterday which was a bit better but still way too tight a fit to readily attach/remove.

ggilestro, 2020-10-08:
You should expect hardly any friction. Perhaps try printing with a smaller nozzle setting (0.1 uM).

[image]

rjs360, 2020-10-08:
Wow. That’s a big difference from the fit i am getting.  I will try the 0.1um setting but wondering now if one of my STL files may be a different version from what you show?

I exported the OnShape files as STL’s, is there another repository of the new SD module files that I could try?

Ruud

ggilestro, 2020-10-08:
[This](https://cad.onshape.com/documents/176e8d4c540a50da42228c84/w/6424211cdb0e416f2b1ce2a2/e/b17ba6f469b074872a2a1791) is the STL you should be printing.

rjs360, 2020-10-08:
Yes, that’s the one I used.
I will try some things and let you know if I find out what’s going on.
-Ruud

rjs360, 2020-10-09:
Looks like printing at 0.1mm layer height is a definite improvement on my end.
Not quite as smooth as your gif suggests, but I may be over-extruding just a bit (that may also explain why 0.2mm layers are making it almost impossible to get on manually without breaking things). I can manually secure and remove the covers now. Thanks Giorgio!

closed 2020-10-09 by rjs360