gilestrolab/ethoscope #136: Are "gaps" in data normal?

IllustratedMan-code, 2020-10-26:
I have several gaps in my data as evidenced by the tile plot below. Is this normal? Can I consider it as 0 movement?
Thanks.

[tileplot.pdf](https://github.com/gilestrolab/ethoscope/files/5440149/tileplot.pdf)

ggilestro, 2020-10-27:
No, that does not look normal. without looking at the database file this looks like it could be something on rethomics side too. @qgeissmann do you have any idea?

qgeissmann, 2020-10-27:
Hey, this basically means no data is available (`NA`). 
A priory, that could happen in one of several situations:

* the animal died/stopped moving for extended duration
* the animal escapes
* the animal is not detected, occluded, or leaves the region of interest

Using the snapshots embedded in the db file can help to debug these things.
I hope it helps!

IllustratedMan-code, 2020-10-28:
The animal of interest are ticks which have long periods of inactivity. How extended of a duration would cause the gaps? Thanks

IllustratedMan-code, 2020-10-28:
I am also using the custom ROI builder if that matters

ggilestro, 2022-01-22:
Closing for lack of further activity.

closed 2022-01-22 by ggilestro