gilestrolab/ethoscope #137: Videos>Start streaming issue

rjs360, 2020-11-14:
Not sure if this is a new issue, but I couldn't find a previous report that was similar:

When I start streaming (not tracking) video from my three ethoscopes, I only receive a blank square as feedback, and when I then stop the streaming, it asks if I want to stop tracking (?), and if I confirm, the page hangs on "stopping streaming" (plus a blue button stating "stopping"). When I close this streaming tab, the ethoscope is still listed but as "stopping", and no stop/start icon next to the name of the ethoscope.

Also, this process does not seem to throw an error in the log?

When I instead initiate tracking, things seem to be working better.

Running latest Node and Ethoscope images, Manjaro Linux.
Any insight would be most appreciated.

rjs360, 2020-11-14:
For good measure, here are some screen shots of what happens:

Initial list of devices:
[image]

After starting a stream on one device:
[image]

After stopping the stream tab on that device:
[image]

After refreshing the tab (after stopping the stream):
[image]

After closing the (now frozen) streaming tab:
[image]

ggilestro, 2020-11-16:
Hi - can you confirm which versions of the software you're running, please?
This definitely works with the latest dev but I'll probably merge to master by the end of November anyway.

rjs360, 2020-11-16:
Hi Giorgio,

That would be for the node: manjaro-ethoscope-20.0.1-stable-minimal-x86_64.iso
And for the ethoscopes: 20201110_ethoscope_000.zip

Ruud

ggilestro, 2020-11-16:
What version are you running on the node? You can go to the updater page and check the current version.

[image]

rjs360, 2020-11-16:
Sorry that’s:[dev] 20a8ed… (20-10-29 20:01:46)

-Ruud

[image]<https://user-images.githubusercontent.com/696901/99291268-73708b80-2837-11eb-983e-e132d49a7257.png>

rjs360, 2020-11-17:
Hi Giorgio, thanks again for the follow-up. This now seems to be resolved (not sure how exactly though?).

closed 2020-11-17 by rjs360