gilestrolab/ethoscope #138: ethoscope does not appear in the web interface

Nocturtle, 2020-12-15:
Hello,

I am running the 20201127_ethoscope-node_pi4 image on a RPi4b and 20201126_ethoscope_000 on a RPi3b. Both devices are connected to my school's network (the node via ethernet, the scope via wifi) but the ethoscope is not getting detected by the node in the interface. I have pinged the ethoscope from the node and vice versa with no packets lost. I am at a loss of how to troubleshoot this further. Any advice would be greatly appreciated.

Thanks,
-Daniel Harvey

pepelisu, 2020-12-16:
Hi Daniel,
It could be that your university network is using a firewall and therefore the port 5353 (required to use avahi) is closed. 
Ethoscopes are discover automatically by the node using avahi/bonjour protocol.
The recomended set up is to have a private network without internet access for the ethoscopes and node. Only the node is connected to internet through university network, that way:
1. Ethoscopes are at least minimally protected to unwanted access. 
2. Ethoscopes can work without port restrictions and firewalls 

Other ports needed for the sytems to work are 9000 and 8080 ( I think)

Nocturtle, 2020-12-17:
Thank you for this info. I will get in touch with my IT to see if these ports are open on our dedicated IoT network.

ggilestro, 2020-12-17:
@Nocturtle you can try running the following command on the node: `avahi-browse --all`
You should get the list of all the discoverable zeroconf devices on the network and if you see no output you'll know it's a firewall issue.

Nocturtle, 2020-12-18:
@ggilestro After first running that command I got a message that said "Failed to create client object: Daemon not running". Manually starting the daemon with 'avahi-daemon -D' seems to have fixed the issue. I can now see the ethoscope in the interface when it is plugged into ethernet, but not on wifi. Hopefully I will hear back from IT about getting those ports open. Thanks!

closed 2022-01-22 by ggilestro