gilestrolab/ethoscope #140: optomotor module demo works on LEDs but not on motors

rjs360, 2021-01-06:
Curious if there is a record of this issue:  I am testing the (now slightly aging) optomotor module currently, and while the (push button induced) "DEMO" will turn on LED's, my dc motors are not turned on.  Similarly, when connected to an Ethoscope, it will run the tracking program, and turn on LED's during the initial demo that is run prior to the start of tracking, but it will not do anything to the motors. 

I double/triple checked the wiring as described in this module's build instructions, and even used different connectors (i.e. the white 6-way female connectors are hard to solder, but the issue persist using individual basic arduino wires) to connect the arduino micro and the TLC board.  My motors seem fine, I can make them run using a basic 2-AA battery supply, and are in fact the same motors that I can make work just fine with the new modular sleep deprivator module connected to ethoscopes. 

One note is that the TLC board I am using (while same model no.) is not the exact Adafruit product, as they were backordered at the time. Their board and mine are identical superficially though..  Do I perhaps need to change any settings in the arduino code?

Thanks for any insight.

ggilestro, 2021-01-06:
Make sure you give your module enough amperage. Motors need quite a bit of current to spin and they won't move if you give them less than 1A. Let me know.

rjs360, 2021-01-06:
Hi Giorgio,

I tried powering the TLC5957 board with 4 batteries and omitting the arduino micro altogether, but that doesn’t change things, LED turns on just fine, motors are still…. Four (or even two) batteries by themselves power the motors fine.  Is this TLC board limiting the current it can supply to the motors?

-Ruud

ggilestro, 2021-01-06:
Not sure what to say. I have no experience with non-Adafruit boards and I don't think we've seen this problem before.

rjs360, 2021-01-06:
Gotcha. I will see if I can test an Adafruit board.
If not I will turn to the new SD module.

rjs360, 2021-01-12:
As a follow-up, I tested an original Adafruit board (TLC 5947) and am getting the exact same results.  With the 1.15Kohm resistor in place, LEDs will light up when the demo is run (via the push-button bypass), but no response from the motors. Again, these motors run fine using 4 AA batteries in series (e.g. like the new modular SD module).  Perhaps I will just use this with the LED functionality, but it is odd, no?

ggilestro, 2021-01-13:
Odd indeed.

closed 2022-01-22 by ggilestro