gilestrolab/ethoscope #142: Change size of object to track

tshaible, 2021-01-14:
I am trying to increase the size of the tracked object, as I work with Monarchs rather than flies. 

I have included the following line of code in an attempt to do so, but it doesn't work: 

self._object_expected_size = 0.05 # proportion of the roi main axis
self._max_area = (2.5 * self._object_expected_size) ** 2

Any help would be greatly appreciated

ggilestro, 2021-01-18:
Changing the expected size should suffice

https://github.com/gilestrolab/ethoscope/blob/0521648f33103b8055342aea3ef37c06fde150a1/src/ethoscope/trackers/adaptive_bg_tracker.py#L249

If you send me a short video (few minutes) I can try to have a look.

tshaible, 2021-01-20:
> *External Email*

day8.avi
<https://drive.google.com/file/d/1jLeAMAFmzBl4fMZ9g4b-0EveUzQANfpH/view?usp=drive_web>
 moreblocks.png
<https://drive.google.com/file/d/1cgBkUu8_QDfVmoLfNTSSzqyFU0M5nkRn/view?usp=drive_web>
 trimmed.mp4
<https://drive.google.com/file/d/1bD0EvVNj4-undaYpFVgOlcetNq32dRvl/view?usp=drive_web>
Hello,

I have finally been able to get the size change to work, although even when
I've changed the expected size to be quite large, there are still many
small objects tracked (see attached video named day8).

I realize there is too much background noise in this video and in the
future we are going to better account for this, but I'm wondering if
something might be done to clean up the tracking? I attached the original
video as well as the roi image I am using as well.

Thanks for any help you can provide,
Tierney Shaible

tshaible, 2021-01-29:
Just wanted to follow up on this matter to see if you might be able to
provide any help or direct me to someone who can help.

Thanks!

Tierney