gilestrolab/ethoscope #143: Total Beginner's Question: Node Interface

zzaidi148, 2021-01-26:
To preface, I am a total beginner when it comes to coding, linux, and RPI's in general. My experience has always been in biology rather than computers but, after seeing this project, I am hopefully planning on changing that. 

**My Question:** 

So I have flashed the appropriate node OS onto my Samsung 128gb sd card and inserted it into my RPI4 to act as a node, based on the guidance given in the Ethoscope User Manual Notion Page. The unit boots up and shows me a terminal that requests for the ethoscope's login information. I was assuming it would boot up to the Node Interface, however, after trying "root" as both the username and password, I am lost in terms of how to get to the Node's UI. I currently have my node plugged into a D-Link 825 wifi router that I believe has been properly configured in terms of IP addresses. I have reserved an IP address for my node, however, not knowing the RPI's hostname, I'm not entirely sure whether it is actually receiving the static IP that is assigned to it. Besides this, I have not touched any of the actual ethoscopes as of yet. Your guidance in solving this issue will be greatly appreciated!!!

 _Attached is what I currently see on the node._ 

[image]

ggilestro, 2021-01-26:
You can login on the node using username `alarm` and password `alarm`.
From there, you can find what is the local IP address using the command `ifconfig`. Look for something like `192.168.1.x`.
Write down that IP address and then from another computer on the same network you can point your browser to it to see the node interface.

zzaidi148, 2021-01-26:
Thank you @ggilestro!! I have successfully been able to reach the node interface. I am now wondering how I can do an update of my system. I do not believe I am allowed to be hard-wired via ethernet into my university's internet access points, so I am wondering if there is any way I can connect to the wifi from the alarm@node (node) terminal. Also, if this is just a one-time thing for a software update, I can maybe update the device from home then bring it back to the university. Please advise. I am very much grateful for your prompt responses!!!

ggilestro, 2021-02-03:
You can certainly update from home. You will need to be connected to the internet in order to be able to update.

closed 2021-02-03 by ggilestro