gilestrolab/ethoscope #144: A date and time range in which the device will perform

posttenebre, 2021-01-26:
I am wondering what the date and time range format is for this section. The url given is no longer active, and the code does not mention it either. Thanks in advance. 

[image]

posttenebre, 2021-01-28:
I took a closer look at the code and determined that it should be something along the lines of
`2020-01-28 20:00:00 > 2020-01-30 20:00:00`
Am I correct?

ggilestro, 2021-01-28:
Yes, you are correct. Apologies for the slow reply.

closed 2021-01-28 by ggilestro

posttenebre, 2021-01-29:
No worries. Should have looked deeper in the code initially. Will the manual for using the ethoscopes be available again soon?