gilestrolab/ethoscope #145: Ethoscope going into emergency mode

ThHoler, 2021-02-01:
Hello, 

Recently our ethoscopes have been having a new issue. Once an experiment is done and we end the acquisition, the ethoscope sometimes disappears entirely from the list of available machines on the Platform Overview page. This happens without us actually shutting the ethoscope down. 

We tried connecting a monitor to the ethoscope using an HDMI cable and this is what we saw :
[image]

So far the only solution we have found to that issue was to reset the SD card and flash it again, which is quite bothersome considering it happens somewhat often. What could be the reason for that issue and the solution to it ?

Thanks in advance.

ggilestro, 2021-02-02:
This is a sign of failure of the SD card. Unfortunately, the failure rate in SD cards is the weakest aspect of the platform at the moment and depends largely on the brand of SD cards one uses. We have had the best results with Samsung cards in the past.

ThHoler, 2021-02-02:
Thank you for the answer.

We have been using Samsung 64 gb cards. Is there any way to mitigate the risk of SD card failure ? I suppose it isn't possible to use a USB stick to host the software for instance ?

ggilestro, 2021-02-03:
I have made a change to the code that removes some heavy writing activity on the SD card upon completion of the experiment. Perhaps this was the issue. You may want to try and update your ethoscopes now.

ggilestro, 2021-04-15:
Closed for lack of activity

closed 2021-04-15 by ggilestro