gilestrolab/ethoscope #147: Choice of N20 motors for optomotor module

antortjim, 2021-02-07:
Dear Giorgio and ethoscope community

**TLDR; Criteria to choose N20 motors and how to debug motors failing through experiment.**

I wanted to ask what N20 motors do you recommend when building an optomotor. I saw in your documentation here https://www.notion.so/d3e5858fedf14e19a235cae29c0f3f2a?v=40f684f220f44e25a292d85d98f0d316&p=a478fcbff07c469db1b531a7f50d336d that you currently recommend this supplier on AliExpress https://www.aliexpress.com/item/32778906874.html?spm=a2g0s.9042311.0.0.27424c4dk02sF5

We in the lab have been using this one https://nbleisonmotor.en.alibaba.com/product/60151519001-802733618/12mm_3v_6v_12v_1_10_to_1_1000_ratio_small_N20_Micro_dc_gear_motor_for_robot_and_door_lock.html?spm=a2700.shop_pl.41413.10.69da4f14HobF1Y (model N20AGB298-09168) with not ideal results. We chose this model because it has an RPM similar to what we want (~1 rev / second i.e. 1 rev / stimulus), but we are not completely satisfied with the results. We experience the motors to be not reliable enough (sometimes they stop in the middle of the SD). I suspect it is either due to 1) the gears of the motor getting stuck in the 3D printed material and/or 2) a too low torque that causes the motor to sometimes not be able to overcome the weight of the tube or some irregularity.

I wanted to know why do you recommend this supplier in particular, and whether you know of any supplier in Europe (we are located in Belgium, but I am sure people elsewhere may be interested also in alternative suppliers) since shipping from China is so expensive and slow (40 euros delivery and 2 months wait time). For instance those from Amazon Germany and US:

* https://www.amazon.de/-/en/elektromotor-Miniatur-DC-Getriebemotor-Drehzahl-Motor-15-1000rpm-Zahnrad-Gleichstrommotoren/dp/B08599WJHH/ref=cm_cr_arp_d_product_top?ie=UTF8&th=1#descriptionAndDetails

* https://www.amazon.de/-/en/Reduction-300RPM-Gearbox-Replacement-Engine/dp/B01BBSXDJ4/ref=pd_bxgy_img_2/260-2387255-1582513?_encoding=UTF8&pd_rd_i=B01BBSXDL2&pd_rd_r=fb0eec9e-ab7e-4c7b-822b-7bf1dba402c7&pd_rd_w=Y9EDD&pd_rd_wg=hx8tE&pf_rd_p=4490d326-3239-499b-8c8a-0dd8d6ad582a&pf_rd_r=HT9CX5JRXETR6XDGPWG7&refRID=HT9CX5JRXETR6XDGPWG7&th=1

* https://www.amazon.com/CNBTR-GA12-N20-Miniature-Electric-Gearwheel/dp/B06XWM2Q2B/ref=pd_day0_hl_60_2/138-5052998-6005465?_encoding=UTF8&pd_rd_i=B06XWM2Q2B&pd_rd_r=f02fa6f5-988a-48bd-b8d2-ac19ec871120&pd_rd_w=ba4jI&pd_rd_wg=62kKf&pf_rd_p=0501877d-5f8c-4ec8-9861-e0476eecc53e&pf_rd_r=K95NKQR1TY8TVGYF7JC9&psc=1&refRID=K95NKQR1TY8TVGYF7JC9  

This last one was mentioned to me by @budhaChowdhury We had a chat last week and we agreed it would be great if we could set up a FAQs site on hardware / software debugging.

I find it extremely hard to choose a supplier because the specifications of the motors are almost always missing important info. For example, the torque is frequently not specified!

Thank you in advance!!!!

Kind regards,
Antonio

ggilestro, 2021-02-07:
We have experience only with N20 motors bought on Aliexpress. In total, we bought from three different stores buying 100, 54, and 180 motors respectively and we never really noticed any difference in performance. The motors we've used are all rated between 200 and 300 RPMs. The last store we placed an order from is [this one](https://www.aliexpress.com/item/33059907659.html?spm=a2g0s.12269583.0.0.15d27711Q7wr92). 

Getting components from China takes 2-3 weeks for the UK with the cheapest shipping option - or at least that was the case before Brexit. In our experience problems with the motors are of two types: issue with the power supply or soldering on wires coming undone.

antortjim, 2021-02-11:
Thank you @ggilestro ! We will place an order from that provider too.
We experience quite frequently that motors will move at the beginning of the experiment but then stop moving after a while (within a few hours or even less than 1 hour). We think it's caused by:
1. Eventually the motor and the tube entering a situation where they get stuck with some irregularity on the tube's surface. Or maybe gravity eventually causes the ethoscope to settle in such a way that some motors are stuck.
2. The electronics (either Arduino chip or TLC5947) stop responding properly, maybe because a buffer is full, or maybe because a cable as you said, is not properly connected.

I saw this week an optomotor that worked fine at the beginning of the experiment. 2 days later, no motor moved when commanded to do so using this command and the attached script

`python test_motor.py -m 1 2 3 4 5 6 7 8 9 10 -d 2 -w 2`

But unplugging and replugging the USB cables (Arduino->RPi and TLC5947->RPi) was enough to get that command to work again. So somehow just 'rebooting' or resetting these two chips was enough. Which inclines me to think option 2 is a possible culprit indeed.
[test_motor.zip](https://github.com/gilestrolab/ethoscope/files/5964773/test_motor.zip)
Your PSU for the motors is the RPi, correct? Or do you use an external PSU?

I discovered AdaFruit is making motors that report whether they actually move or not. Maybe they are overkill and too expensive, but good to know they exist! https://www.adafruit.com/product/4640

ggilestro, 2021-02-11:
Our PSU for the motors is *NOT* the rPI. We use two USB cables. One goes from the Arduino to the rPI and it is used for serial communication. One goes from the PCB to a separate power supply and it is used for powering the motors. The motors need 200-400 mA each if I remember correctly so it is not wise to power them via the rPI.

Also, we now transitioned fully to the new electronics. A description of it and the Gerber files to order the PCBs are on the manual webpages.

closed 2021-03-19 by antortjim