gilestrolab/ethoscope #149: Database and Backup problems

pgorelik, 2021-03-06:
Hello,

We have been unable to get data off our Ethoscope, and I suspect that the database is not being properly created on the device.

First off, on the node web UI, the save location is shown as `/None` but this might be unrelated.

When I run `backup_tool.py` I get the following results:

<pre><code>
INFO:root:Backup done for for device  007a915c416b4a86be04b0049428a9bb
INFO:root:Pool mapped
INFO:root:Joining now
INFO:root:Backup finished at t=1611959278
ERROR:root:Could not generate backup path for device. Probably a MySQL issue
ERROR:root:Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/ethoscope_node-trunk-py3.7.egg/ethoscope_node/utils/device_scanner.py", line 496, in _make_backup_path  buffered=True)
  File "/usr/lib/python3.7/site-packages/mysql/connector/__init__.py", line 173, in connect  return MySQLConnection(*args, **kwargs)
  File "/usr/lib/python3.7/site-packages/mysql/connector/connection.py", line 102, in __init__    self.connect(**kwargs)
  File "/usr/lib/python3.7/site-packages/mysql/connector/abstracts.py", line 735, in connect    self._open_connection()
  File "/usr/lib/python3.7/site-packages/mysql/connector/connection.py", line 252, in _open_connection    self._ssl)
  File "/usr/lib/python3.7/site-packages/mysql/connector/connection.py", line 172, in _do_auth    self._auth_switch_request(username, password)
  File "/usr/lib/python3.7/site-packages/mysql/connector/connection.py", line 216, in _auth_switch_request    raise errors.get_exception(packet)
mysql.connector.errors.ProgrammingError: 1049 (42000): Unknown database 'ethoscope_db'
</code></pre>

I am not sure what to make of the mysql error.

When I ssh into the device, there are .sql files in `/ethoscope_data/backup` but not in `/ethoscope_data/results` If I record a video, however, it seems to go into the appropriate directory.

I have had this same behavior using at least 2 different SD cards. Currently we are running 20201126_ethoscope_000.img, and I burn the images using Balena etcher. However, I experienced the same behavior with at least one other earlier disk image (20191121)

I think I must be doing something silly, but I am out of my depth. We'd appreciate any help.

Sincerely,
pavel

antortjim, 2021-03-19:
Make sure you have the right versions of the ethoscope repository checked out both on `/opt/ethoscope-device` in your ethoscope and in `/opt/ethoscope-node` on your node.

Also, make sure the `mysql` module is version 8 (that's the one I have now and it works). I have experienced weird errors because of this module in the past
Finally, how do you run the web UI? Do you have the systemd service running?

ggilestro, 2021-03-19:
This actually looks like a problem on the node side, not on the ethoscope side.
Make sure you are running the latest version of the node.

pgorelik, 2021-03-24:
Thanks very much for your feedback! I updated the node to OS to Manjaro with the latest SW and it fixed the problem.

closed 2021-03-25 by ggilestro