gilestrolab/ethoscope #153: New Scope asking for login / password

jbrownlieBNE, 2021-04-13:
Hi all,

I have built 4 new scopes - the first two ran ok, but the next two are asking for a login and password.

I can see all 4 devices have connected to the network and all have unique IP addresses assigned by the router. The Node can see the first two scopes, and I can interact with them via the webpage. 

I have tried to manually add the remaining scopes but receive an error from the Node stating "The following entries could not be added: IP xxx"

I have also run a Raspberry Pi Desktop OS to confirm the units aren't faulty, and all appears fine. The units connect to HDMI / USB outlets and run the OS happily.

I have tried to use the other SD cards that are running the functional scopes, but these don't seem to want to connect to the node. It doesn't ask for a login but stalls after the rainbow screen.

All four SD cards have been burnt using the latest image file (20201126_ethoscope_000), and all within 24 hours of each other.

ggilestro, 2021-04-15:
Sorry, I am a bit confused as to what is happening exactly. What is the problem? The ethoscopes do not boot? 
The [rainbow screen](https://lifehacker.com/what-the-raspberry-pis-rainbow-boot-screen-and-rainbow-1768470271#:~:text=Speaking%20of%20power%20supply%20problems,red%2C%20it's%20over%2Dtemperature.) indicates a problem either with the power supply or with the PI.

jbrownlieBNE, 2021-04-15:
The scopes boot up normally, I can a bunch of commands being executed but then it stops and asks for a login and a password.

I have tried using ETHOSCOPE_1234 but it doesn’t work.

As I noted earlier I can see that the scopes connect to the WIFI network, but are not connecting or interacting with the node software

Cheers
Jeremy

jbrownlieBNE, 2021-04-15:
[image]

jbrownlieBNE, 2021-04-15:
[image]

jbrownlieBNE, 2021-04-15:
and now the scopes that were connecting are no-longer connecting. They too are now asking for login and password.

jbrownlieBNE, 2021-04-15:
[image]

jbrownlieBNE, 2021-04-15:
[image]

antortjim, 2021-04-16:
Hi @jbrownlieBNE 
In [the supplementary material S3 of the Ethoscope PLOS One publication](https://doi.org/10.1371/journal.pbio.2003026.s004
) under section **Restarting services** > **In a device** it is mentined that the user and password of the ethoscope image is **alarm** and **alarm**. You can also try **root** and **root** to login as administrator.

It is very common to find the root/root or admin/admin combinations in factory default software like this

closed 2022-01-22 by ggilestro