gilestrolab/ethoscope #154: Exception: Async database writer has stopped unexpectedly

posttenebre, 2021-04-21:
I get this error after a few hours of running the ethoscope.

Software version: (2021-04-02 11:07:59)
rasberry pi 3 b+
mAGO with new SD module

`Sorry there has been an error: Traceback (most recent call last): File "/opt/ethoscope-device/src/ethoscope/web_utils/control_thread.py", line 485, in run self._start_tracking(cam, result_writer, rois, TrackerClass, tracker_kwargs, hardware_connection, StimulatorClass, stimulator_kwargs) File "/opt/ethoscope-device/src/ethoscope/web_utils/control_thread.py", line 342, in _start_tracking self._monit.run(result_writer, self._drawer) File "/opt/ethoscope-device/src/ethoscope/core/monitor.py", line 139, in run raise e File "/opt/ethoscope-device/src/ethoscope/core/monitor.py", line 131, in run result_writer.flush(t, frame) File "/opt/ethoscope-device/src/ethoscope/utils/io.py", line 535, in flush self._write_async_command(c) File "/opt/ethoscope-device/src/ethoscope/utils/io.py", line 626, in _write_async_command raise Exception("Async database writer has stopped unexpectedly") Exception: Async database writer has stopped unexpectedly
No Debug Image`

posttenebre, 2021-04-21:
I replaced the SD card with a fresh image but the error occurs nonetheless.

posttenebre, 2021-04-21:
WIth the new SD card I also get a new error. The other two ethoscopes have been running for two days without issue. 

`Sorry there has been an error: Traceback (most recent call last): File "/opt/ethoscope-device/src/ethoscope/web_utils/control_thread.py", line 471, in run raise e File "/opt/ethoscope-device/src/ethoscope/web_utils/control_thread.py", line 467, in run cam, rw, rois, TrackerClass, tracker_kwargs, hardware_connection, StimulatorClass, stimulator_kwargs, self._info = self._set_tracking_from_pickled() File "/opt/ethoscope-device/src/ethoscope/web_utils/control_thread.py", line 354, in _set_tracking_from_pickled return pickle.load(f) _pickle.UnpicklingError: invalid load key, '\x00'. `

posttenebre, 2021-04-22:
I replaced the raspberry pi and still got the same error. I'm at a loss as to what might be cause.

ggilestro, 2021-04-22:
This is usually an indication that the SD card is damaged or poor quality. I recommend using Samsung EVO+ 32GB. Every other SD card we tested gives us problems sooner or later.

posttenebre, 2021-04-22:
I am using the samsung evo plus 32GB as described in the instructions. One of the errors occurred after flashing a fresh SD to a new out of the packaging SD. The error has shown up on two separate Raspberry PIs and three distinct SD cards. These were all sharing the rest of the hardware.

posttenebre, 2021-04-23:
After rebooting the node and the backup service all three ethoscopes, including the one causing issues, have worked without problems. I am wondering if the node was somehow to blame, but the ethoscopes should be able to run independently of the node.

ggilestro, 2021-04-23:
Yes, the ethoscope work independently of the node. The node is merely getting the data out of them every now and them so I don´t see how the node could be responsible for this, sorry. I will close the issue for now but feel free to reopen if new information arises.

closed 2021-04-23 by ggilestro