gilestrolab/ethoscope #156: Ethoscope Structure/Build Question

zzaidi148, 2021-04-21:
Hello, 

I hope you are doing well. Just wanted to ask the community about some of your suggestions as to how you build/print your ethoscopes to have them being as stable and robust as possible. To build mine, I actually 3D printed all the parts including the side walls using the STL files from the shared onshape folder. However, as a result of printing the square laterals and observing issues with the thickness of them, I had to use a dremel tool to shave them down to be able to fit into the notches of the upper and lower cases. I was just wondering if anyone could guide me on their experience using acrylic to maintain the stability of the sidewalls or using the other sidewall design provided that appears in an X shape. As a result of my walls' instability, I am having tracking issues and failure to initiate tracking because splicing of the tracking lanes is either overlapping or out of view of the camera. Any help will be much appreciated!

The following are the shapes I am referring to: 

What I've printed and used that works, mostly:
[image]

How this has messed up my tracking so far: 
[image]

What I'd like your opinion on: 
[image]

and this stabilizer: 
[image]

antortjim, 2021-04-21:
Hi @zzaidi148 

Regarding the fitting, yes, it's not easy to fit wall and box together, but that has the advantage of making it a solid junction. It's hard to hit the sweet spot of an easy to make yet difficult to break junction... Someone else please correct me if I am wrong.

Regarding the lighting, what material did you print your walls with? We print ours with white PLA. Did you notice there's one side more glossy than the other? Maybe this glossy side is facing the flies and causing the reflections?
In any case, if these reflections are caused by light in the incubator, I strongly suggest you install a filter on the camera, such that only the IR light at 850 nm arrives to the camera. The result is all your images will look like the D phase (of course the animal behavior is completely unaffected).
This would be a simple workaround to your tracking problems. I can send the reference of the filter if you are interested, as well as the mount you can use to put the filter on (or maybe just stick it to the camera). 

Cheers,
Antonio

ggilestro, 2021-04-22:
I agree with @antortjim that installing an infrared passthrough filter is the best way to solve reflections.
Regarding the walls: we still exclusively rely on laser-cut walls. If you'd rather print, you will have to find what works best for your printer. You may have to modify the files so to print a slightly larger or smaller tab. The stabilizer is meant to be used for laser-cut walls.

Once you find a good fit, you may want to use a bit of glue to hold things tight together. We tend to keep the ethoscopes always in the incubators and hardly ever take them out.

zzaidi148, 2021-04-22:
Thank you both for your responses. I am afraid I was not clear as to my exact issue. I actually did not consider the reflections to be an issue until you both brought it up, so @antortjim I would appreciate it if you could send me the filter information. My main issue was the fact that my tracking lanes were not properly aligned with the tubes in the arena. This means that there is an overlap of tubes in one tracking lane where two flies could be tracked at the same time or that if a fly was slightly out of the tracking rectangle that it would not end up being tracked at all. Here is an example of what I mean. 
[image]

I was just wondering if this could be solved with any structural modifications. Currently,  the walls I had printed were sanded down to fit into the notches, but since this was done by my own two hands, I may have introduced a slight angle that may have compromised the angle of the camera (by virtue of teetering walls)  and the alignment of the tubes within the arena. I also, due to my walls not being stabilized, have been suffering from an inability to initiate tracking in some ethoscopes because there is an area of the arena that is slightly out of the scope of the camera. Usually the system will give me a coordinate that is causing the problem, but due to the obvious absence of a grid, I am unaware of exactly what location it is referring to. I was just wondering if you had any further suggestions on how I could deal with pan or zoom of the camera if using acrylic walls is not the solution. 

Thank you for all your input.

pepelisu, 2021-04-23:
Looking at your image the main problem as you said is the alignment of camera and arena. To improve that, you can print the frame that you show earlier, that would stabilize the whole thing, However that frame is designed to work with acrylic walls and or flat walls, you can use any other material, like wood or similar if you have access to it, the only critical part is that the thickness of the material is 3mm. 
The walls are designed to fit and stay in place by pressure, that is why they are hard to get in place. 

Also looking at the distortion in the image I think you also have the camera a little bit angled inside the head cover, probably is screwed to tight and the camera sensor is angled with respect to the bottom port of the RPi box. Check that the camera is not tilted, and then focus on the position of the walls.

zzaidi148, 2021-04-23:
Thank you @pepelisu! I will try and examine all variables and see how I can ameliorate the issue. I also, on an unrelated note, wanted to ask about the community's opinion on this food bullet arena and if it is useful and available to use with the current ethoscope software. I think it's a brilliant design than can increase the number of flies tracked and make loading them much faster. 
[image]

I am however wondering how you can manage to keep the flies in the lanes. Do you just use cotton to seal the end?

[image]

pepelisu, 2021-04-23:
We  use millipore tape from 3M, but I think  you can also use cotton plugs. The downside of this arena, (at least when I tried it) is that the food dries quicker than in the tubes, maybe because there is more air circulation. At that moment we where using much more IR light that was translated into heat and water evaporation from the food. Give it a try and let us know how it goes.

antortjim, 2021-04-23:
That is a cool arena! Very handy for loading flies, and also to renew their food right? But from what I see, it doesn't allow SD.

@zzaidi148 This is the filter http://leefilters.com/index.php/camera-directory/camera-dir-list/category/infrared

Good luck!

zzaidi148, 2021-04-23:
Hey @antortjim , what does SD mean? Also, thank you for the link.

antortjim, 2021-04-28:
Hi @zzaidi148 sorry for the late reply. By SD I meant sleep deprivation. Because as I understood it, there's no container that the fly sits in other than the arena itself, which means we cannot use the current SD module with this (maybe another SD protocol would be possible)

closed 2022-01-22 by ggilestro