gilestrolab/ethoscope #158: Node can't find running ethoscopes after crash

posttenebre, 2021-05-05:
Node Raspberry Pi 4.
3 Ethoscopes running sleep deprivation experiment. 

Overnight the node stopped working and required a reboot (not sure what happened). The ethoscopes are still running but do not show up in the interface for the node. Rebooting the ethoscopes works in this situation but at the moment that is not possible. I have restarted the service on the node `systemctl restart ethoscope_node.service` but this does solve the issue. Is there something I can do from either the node or the ethoscopes that will allow for recognition without interrupting the experiment? How do the ethoscopes announce themselves to the node?

posttenebre, 2021-05-06:
If I run `arp -a` I get the following:
```
[alarm@node ~]$ arp -a
ETHOSCOPE008 (192.168.1.13) at b8:27:eb:1d:9b:01 [ether] on enp1s0u1
ETHOSCOPE004 (192.168.1.8) at b8:27:eb:62:2f:93 [ether] on enp1s0u1
_gateway (192.168.1.1) at 9c:c9:eb:4f:43:2b [ether] on enp1s0u1
_gateway (redacted) at 08:96:ad:f6:c9:c0 [ether] on eth0
? (192.168.1.7) at b8:e8:56:1d:00:58 [ether] on enp1s0u1
? (redacted) at 24:4b:fe:58:b3:d8 [ether] on eth0
ETHOSCOPE001 (192.168.1.6) at b8:27:eb:f3:8b:ed [ether] on enp1s0u1
```

Which would suggest that the ethoscopes are registered on the network correctly but the node is not finding them. I tried restarting the avahi service on the ethoscopes just in case.

antortjim, 2021-05-10:
Hi @posttenebre 

The ethoscopes announce themselves to the node using the `zeroconf` module in Python.
I don't know the details of how it works, maybe @ggilestro can help there. But studying the `device_server.py` script (which is the entry point of the ethoscope instance running on each raspberrypi), I found the block of code that does this.

https://github.com/gilestrolab/ethoscope/blob/0521648f33103b8055342aea3ef37c06fde150a1/src/scripts/device_server.py#L475

This is why rebooting the ethoscope or restarting the systemd service does the job. Fortunately, you can manually register the ethoscope, without stopping / pausing the experiment, if you just run the corresponding piece of code. I made a script with just this code by copy pasting from the device server script. You can find it attached to the message

[register_device.zip](https://github.com/gilestrolab/ethoscope/files/6451813/register_device.zip)

If it doesn't still work, try running the same script again, but with the following modification:

```
#address = socket.gethostbyname(hostname+".local")
address = socket.gethostbyname(hostname)
```
 
becomes

```
address = socket.gethostbyname(hostname+".local")
```

Hope it's not too late, and if it is, that it helps in the future!

ggilestro, 2021-05-10:
Sorry, I see this only now.  
If you know the IP address of the ethoscope (like in this case) you can bypass the bonjour protocol and add it manually to the interface using the bar under the filter.

[image]

posttenebre, 2021-05-10:
Thank you both. I had never realised that I could add them manually. I should have looked more carefully at the interface. The ethoscopes are now backing up one fly at a time and will take a few hours to collect all the data. Thank you both for your help.

closed 2021-05-10 by posttenebre