gilestrolab/ethoscope #160: ethoscope_device.service could not be found (Ethoscope unable to connect to computer)

tkh32, 2021-08-02:
Hello, 
The ethoscope runs on raspberry pi 4 b. On attaching the ethoscope to a monitor, i can see that it's connected to the wifi and has an ip address which I have noted. However, I am unable to add the ip address on the localhost page. On following the troubleshooting guide, I think the issue I run into is:
unit ethoscope_device.service could not be found.
I have tried installing the ethoscope node again manually on the RPI4 but the issue persists.

Any help is appreciated.

ggilestro, 2021-08-03:
Are you running the ethoscope on a pi4? Using what image? We have not
released an image for the pi4 yet.

tkh32, 2021-08-03:
Is there any way the image could be modified to run on pi4?

ggilestro, 2021-08-13:
Yes, there is but it will require some work and proper testing of the behavioral output, which is not exactly trivial. It is not just a matter of making sure the software runs on the PI4 but also to ensure the biological read-out is shared with the older PI versions. PI4 has a faster clock so a greater frame rate may possibly change things.

closed 2022-01-22 by ggilestro