gilestrolab/ethoscope #161: ethoscope_device.service could not be found (Ethoscope unable to connect to computer)

tkh32, 2021-08-11:
Hello,
The ethoscope runs on raspberry pi 3 b. On attaching the ethoscope to a monitor, i can see that it's connected to the wifi and has an ip address which I have noted. However, I am unable to add the ip address on the localhost page. On following the troubleshooting guide, I think the issue I run into is:
unit ethoscope_device.service could not be found.
I have tried installing the ethoscope node again manually on the RPI3 but the issue persists.

Any help is appreciated.

tkh32, 2021-08-11:
SSH service works between the devices and the avahi-daemon service is also running

ggilestro, 2021-08-13:
Was the ethoscope image generated in the way explained [here](https://www.notion.so/giorgiogilestro/The-ethoscope-60952be38787404095aa99be37c42a27#a346049baadb40738b2d161ac5d5c9f3)?

tkh32, 2021-08-13:
Yes

ggilestro, 2022-01-22:
Closing for inactivity. Issue probably fixed and obsolete by now.

closed 2022-01-22 by ggilestro