gilestrolab/ethoscope #169: Trouble getting connected to UI

roaldarbol, 2021-11-10:
Hi again - and sorry for being a bother!

I'm trying to follow the online documentation to get my Ethoscope set up, but I've got stuck at the node/network; here's some details about my current status:

- I'm using a RPi 4B with the image file from the documentation (https://imperialcollegelondon.box.com/shared/static/x62wmuqgha5r6l656yavxa5e4r8yg1ww.zip). 
- I'm using a D-LINK router. The RPI 4B is connected to the university internet via ethernet, and to the router via ethernet (with a USB-ethernet adaptor). I've configured the network according to the [documentation](https://www.notion.so/The-network-ae7e06a6e938467bbb306ad6b76defcc), and reserved an IP address for the node from the router interface (I've tried with both Mac addresses `eth0` and `enp1s0u1u4`, see image).
- After booting, I am getting prompted for a login and password; I use `alarm` for both following from #143. 
- Running `ifconfig` only shows `172.16.X.X` IPs. And after connecting my laptop to the wireless network I still cannot open any IPs (looked at the various types in the [supplement to the paper](https://storage.googleapis.com/plos-corpus-prod/10.1371/journal.pbio.2003026/1/pbio.2003026.s004.pdf?X-Goog-Algorithm=GOOG4-RSA-SHA256&X-Goog-Credential=wombat-sa%40plos-prod.iam.gserviceaccount.com%2F20211110%2Fauto%2Fstorage%2Fgoog4_request&X-Goog-Date=20211110T141806Z&X-Goog-Expires=86400&X-Goog-SignedHeaders=host&X-Goog-Signature=865fe184cebcb26370dfa08987b2cdd8bd979b610ddd6a1807667c12f9ccafac8b029c106ca09e987f71c09eb71594033ba9ffa9a833ce59b8b556efd4a32b346a7656ae5f7315ca5dc69040bc51f4633518062262e59167cf38b84c012ba21953afb402fc7f6058c011957dd1c91dbf2b362666412191423eba05fcd4393afb1b1462d83217280bab24f2d87b56e162ac530b2a8f3165d3979c9aed538dde2b641b7f10a8e4ef4b52d4eec4ce9fc81f5cc3d933944272ebd6182382c1126214bd815a522c5da813ff2f41a5ec1983bcc38dd380ffe661a1f1a823d13a28b2c9fbf3ed873192c34ce05ffdb34b79a4d2de1dd86799ed1f71bbe60c4660f2bf08) 1.2.4, What is next?)

As far as I can tell, I've followed the documentation quite closely, but I wonder if I'm missing something - are there further configuration steps I should've done? From the documentation it seems that the RPi 4B disk image should be quite plug-n-play. Again, from #143 I wonder if it's something as simple as not being able to connect to `eduroam` and then not being able to update - could that be the cause (and if so, it might be worth sneaking a warning message in somewhere)? 

And again, I really appreciate the work you guys have put in here and that it's not easy to maintain/develop/customer-service open source software whilst also doing your research. I'm just really looking forward for it to work for me! ;-) 

Cheers,
Mikkel

[image]

ggilestro, 2021-11-10:
Is the node running on the PI4? You can check with the following command
`systemctl status ethoscope_node`
which should return something like this
```
● ethoscope_node.service - Ethoscope node server
     Loaded: loaded (/usr/lib/systemd/system/ethoscope_node.service; enabled; vendor preset: disabled)
     Active: active (running) since Tue 2021-11-09 14:06:35 GMT; 24h ago
   Main PID: 2773626 (python)
      Tasks: 146 (limit: 19060)
     Memory: 6.1G
     CGroup: /system.slice/ethoscope_node.service
             └─2773626 /usr/bin/python /opt/ethoscope-node/node_src/scripts/server.py
```
If you plugged the ethernet cable after powering the PI, you may have to restart the node with the following:
`sudo systemctl restart ethoscope_node`

roaldarbol, 2021-11-11:
Did some troubleshooting with the always helpful [André](https://github.com/amchagas), and it turns out that ***`eduroam` is so restrictive that when it notices a broadcasting device (router), it kicks us off***; that should be explaining the resulting `172.x.x.x` IPs. So just not connecting the RPi to the internet, but only to the router does the trick and the WebUI now turns up! :-D 
As it means that the RPi is not connected to the internet, I'm wondering if there is a way to manually log in to a wireless network with login credentials from the RPi (maybe with `wpa_supplicant.conf`)? An alternative we talked about was to set up yet another Pi as a wireless router (e.g. with [RaspAP](https://github.com/RaspAP/raspap-webgui)).

roaldarbol, 2021-11-11:
> Is the node running on the PI4? You can check with the following command `systemctl status ethoscope_node` which should return something like this
>
> ```
> ● ethoscope_node.service - Ethoscope node [...]

And that was a good suggestion, it now **does** show up as running. I do get a `WARNING`:
```
node python[303]: WARNING:root:Cherrypy version is bigger than 9, we have to change to cheroot server
``` 
Not quite sure what to make of it.

ggilestro, 2021-11-11:
You can ignore the warning about cherrypy. It is just a warning and it's normal. Can you now access the node UI from another computer on the network?

Regarding the network configuration: I do not understand what is the role of eduroam in all this. As far as your university is concerned, the node is just another computer connected to the network via ethernet cable. The router should *not* be exposed to the internet or to the university network: no uni will ever allow that.

roaldarbol, 2021-11-11:
Okay, good to know. Yes, I can access the UI from my laptop so all good. 👍 

I'm not 100% sure about what's going on. And you're definitely right - now I just tried plugging in the eduroam ethernet cable too and things continue to work (although I'm not getting internet connection, probably as I've not supplied any credentials).
So now my setup is completely identical to yesterday, but works - so maybe something in the order of things was off. Maybe it's because the router was daisy-chained to eduroam through to router when being set up... anyways, it works now. 

Re:
> I do not understand what is the role of eduroam in all this. As far as your university is concerned, the node is just another computer connected to the network via ethernet cable.

We've had a ton of truoble getting our RPi's to reliably connect to eduroam, so they're not *just* another computer, though that should indeed be the case.

roaldarbol, 2021-11-11:
Out of curiosity, how do you guys connect the Pi to the internet? Is just plugging in an ethernet cable enough?

closed 2021-11-11 by roaldarbol

ggilestro, 2021-11-11:
We do not use the PI4 as a node. We have more than 100 ethoscopes regularly connected and working so we need a beefy computer to act as the node.

roaldarbol, 2021-11-11:
Ah, fair enough - that makes a lot of sense! @qgeissmann I take it you might be doing/have done this? Any clues? :-)