gilestrolab/ethoscope #170: Ethoscope on RPi's other than 3

roaldarbol, 2021-11-18:
As I don't have any RPi 3's laying around and they're sold out most places due to shortages, I decided to try using the brand new RPi Zero W 2 which has the same chip as the RPi 3. I flash the `20201126_ethoscope_000.img` with RPi Imager, and I think that it should be compatible in principle, but I get stuck at the rainbow screen (which looks perfectly normal, so it shouldn't be a power issue). Any ideas whether this could be remedied by a few tweaks to the image?

Similarly, as the Zero didn't work, I tried my RPi 4 2GB, but alas, that didn't work either.  On the first boot I got a few lines rolling across the screen, but then the monitor just goes to sleep straight away - I don't even get to a rainbow screen. And on subsequent boots it doesn't even show any text anymore. 

I think it would be especially promising to make it work for the Zero 2 because of its smaller footprint and power consumption, so would be awesome if you have any ideas on how to make it work! :-)

ggilestro, 2021-11-18:
I have no experience with the Zero W2. You may want to browse through [these forum pages](https://archlinuxarm.org/forum/viewtopic.php?f=9&t=15633) to see if you find anything relevant. 

The PI4 should work in principle but I am told the `python-picamera` module will not compile on pi4 from arch, so you will not be able to run the ethoscope without fixing that.

I am afraid for now we can only support ethoscopes on pi3/2 and the node on pi4/3.

roaldarbol, 2021-11-18:
That's fair. I'll give it a go. Seems it's probably the `kernel.img` that needs to be updated ([discussed here](https://raspberrypi.stackexchange.com/questions/19354/raspberry-pi-with-boots-up-with-rainbow-screen)). Which `kernel.img` have you used, or where did you get it? Is it built on top of Raspbian, Raspberry Pi OS or Raspberry Pi OS Lite - or something completely different? 
Or maybe a simpler question - which files HAVE you changed in an existing image that I would need if I adapt a new disc image?

ggilestro, 2021-11-18:
The user manual has [a full description ](https://www.notion.so/giorgiogilestro/Create-a-new-ethoscope-SD-image-fe83714a43584c61960151745a2a9d99)of how the SD images were generated, step by step.

roaldarbol, 2021-11-18:
Briliant, thanks! I've had to order a card reader for the Pi (my only Linux devices around) but will keep you up to date on any progress! It seems that it should work with the `rpi4` image... fingers crossed.

roaldarbol, 2021-11-25:
I've managed to create a new disk image that runs successfully on RPi Zero 2, and probably also on the RPi 4 if anyone would want to use that. I haven't managed to see it in the UI yet, but have the same issue with the RPi 3 (I got my hands on one) - will make a new issue for that.

ggilestro, 2021-11-25:
Thanks. If you could share the instructions of how you created the SD image it would actually be even more helpful than the image itself.

roaldarbol, 2021-11-25:
I've actually made public comments with [hypothes.is](https://hypothes.is/), so you can check them out right on the installation page.

roaldarbol, 2021-11-29:
So it seems that there's an issue with `opencv` before things are up and running on the RPi 4 ArchLinux image. (see [issue](https://github.com/gilestrolab/ethoscope/issues/171#issuecomment-980308337)). I think it's just not installed. I'm expecting it should be possible to just [install it with `pacman`](https://stackoverflow.com/a/45187495/13240268).

ggilestro, 2022-04-28:
Given the PI3 shortage, I went back to this. I think I have fixed in the following way.

1. Install archlinux for arm architecture 64
2. Install python-pycamera-git using the fixed PKGBUILD below:
```
# Maintainer: Grey Christoforo <first name at last name dot net>

pkgname=python-picamera-git
_module='picamera'
pkgver=r929.7e4f1d3
pkgrel=1
pkgdesc="A pure Python interface for the Raspberry Pi camera module."
url="http://picamera.readthedocs.io/"
depends=('python')
makedepends=('python-setuptools')
provides=('python-picamera')
license=('BSD')
arch=('any')
source=("$_module::git+https://github.com/waveform80/picamera.git")
md5sums=('SKIP')

pkgver() {
  cd "$srcdir/$_module"
  printf "r%s.%s" "$(git rev-list --count HEAD)" "$(git rev-parse --short HEAD)"
}

package() {
 cd "${srcdir}/${_module}"
 python setup.py install --prefix=/usr --root="$pkgdir" --optimize=1 --skip-build
}

build() {
    cd "${srcdir}/${_module}"
    python setup.py build
}
```
3. Install ethoscope-device as normal from AUR but revert all changes in /boot/config.txt before boot. The file should only contain the following line
```
enable_uart=1
```

The resulting working image can be found [here](https://imperialcollegelondon.box.com/shared/static/ugqbbmienvt8v7veyu7lmfjcsojp0fss.zip). I have NOT tested it on actual flies. It is unlikely but possible that an increase in the PI4 speed may change the actual behavioural data and will need to be normalised.

ggilestro, 2025-02-18:
The ethoscope software is now working well on [PI4 too](https://www.notion.so/giorgiogilestro/The-node-9cd2d45f3eae46c7a0751c500fa11bac#9422e2b05b794e5786a8d4b0632216a9), using the picamera2 stack and a debian based image. A 40GB image can be downloaded from [here](https://imperialcollegelondon.box.com/shared/static/ugqbbmienvt8v7veyu7lmfjcsojp0fss.zip).

closed 2025-02-18 by ggilestro