gilestrolab/ethoscope #172: Ethoscope display issues with Raspberry PI 4

DaveMelkani, 2022-07-19:
Hey Ethoscope Team,

I purchased all the materials for an Ethoscope, but was only able to procure a Raspberry Pi 4. I used the provided beta image and was unable to find my Ethoscope on the Node website. I get a GPIO error when I tried displaying the ethoscope PI on a monitor using the provided beta image. How should I proceed further?  

[image]

pepelisu, 2022-07-20:
Hi,
The first time that you start the image it makes a series of tasks and sometimes a restart is required. One way to do a soft restart is to ssh into the ethoscope.  To do that:
1. Check the IP assigned to the RPI in your router, the one that is providing the Ethoscope_wifi signal. 
2. Open a terminal in the node and just write: ```ssh alarm@ip_from_the_step_1 ``` for example ```ssh alarm@192.168.0.32```
3. Once you are logged in via ssh, you can then do sudo restart and the RPI should restart. 
4. If something of the above fails, you can try a hard reset, just disconnect the RPI from the mains.

ggilestro, 2022-07-20:
See the section called `Ethoscope does not show up online`  on the [manual](https://www.notion.so/giorgiogilestro/Software-issues-7de6d3569a014c4eb7f14bb8d5a77d4e).

DaveMelkani, 2022-07-21:
Hey Ethoscope Team,

I will try this, but I also had another question. When I tried creating a new ethoscope SD image provided. Building the partitions went well, but in the “SSH session inside the PI” steps, I got stuck on “yaourt -S ethoscope-device.” When I tried running this command, I got an error with picamera package() where it says “ERROR: A failure occurred in package(). Aborting…” I was wondering what I can do about that?

ggilestro, 2022-07-21:
Sorry, I am confused now. Did you use the SD image found [here](https://github.com/gilestrolab/ethoscope/issues/170#issuecomment-1111990628)? Or did you build the image from scratch? If you build the image from scratch you need to change the PKGBUILD to fix the compatibility issue between `picamera` and pi4, as explained in that comment.

if `picamera` is not properly installed the ethoscope will not be working and the node will not be able to find it on the network.

DaveMelkani, 2022-07-25:
Okay, so I assume that I have to copy and paste everything below the lines “pkgname=python-picamera-git" in the #170 issue. I was wondering how do I get to that part where I would have to copy and paste that stuff? Right now in Arch Linux, I at the part where I made a directory /tmp/yaourt and cd into there. How do I change the PKGBUILD, so I can copy and paste the stuff in #170?

DaveMelkani, 2022-07-25:
Okay, I did this where I changed the PKGBUILD to match the one found in #170 and on the last part it says “Install ethoscope-device as normal from AUR but revert all changes in /boot/config.txt before boot. The file should only contain the following line
enable_uart=1" which I did. When I then tried running yaourt -S ethoscope-device, I got this new error. What should my next steps be in proceeding? 

[image]

ggilestro, 2022-08-01:
Hi Dave - please just use the image that you find linked at the end of that comment. No need to recreate an image from scratch.

closed 2022-08-01 by ggilestro