gilestrolab/ethoscope #173: Issue with Tracking

nwoodweb, 2022-07-29:
Hello,
My name is Nathan Wood and I am currently an incoming graduate student undergoing a rotation as part of Dr. Wanhe Li`s group as Texas AM University. My tasks are to assemble and configure apparatus, including an Ethoscope setup.

However, I do have issues with tracking through the ethoscope web interface. 

## Installation
I have sucessfully setup a node with 5 devices and router, in accordance with the instructions provided [here](https://www.notion.so/giorgiogilestro/Ethoscope-User-Manual-a9739373ae9f4840aa45b277f2f0e3a7). 

I have also been able to successfully record video on each device, and have been able to use `ffmpeg` to concatenate and convert h264 files and use `rsync` to backup to an external drive.

Each device can be accessed via `SSH` either as root or as alarm

## Issue 
The primary issue I have now is the use of the tracking feature, as activated from the node webserver.
The tracking had been started using the options `DefaultStimulator` and `SleepMonitorWithTargetROIBuilder` 

Ethoscopes 3,4, and 5. Fail to begin tracking, and throw an exception statement error stating
``` 
There should be three targets. Only X objects have been found
```
[image]
[image]
This error seems to continue regardless of the arena being fully populated, partially populated, not populated, or removed entirely

I have also recieved  a single instance of an exception detailing:

```
Something must be wrong since all target should have the same size.
```

[image]

This exception was thrown despite having the entire arena populated with fly tubes. 

Ethoscopes 1 and 2 have another issue, where there are no error messages at all, and the webserver page is entirely blank, and I must restart the device by unplugging it in order to circumvent this. Unfortunately I was negligent and failed to create a copy of `dmesg` and `journalctl` output  Ethoscopes 1 and 2. However, I did not see anything anomalous on the hardware side of things. 

Unfortunately I have been unable to diagnose these issues and I was curious if anyone has had experience with a similar tracking issue. Any assistance would be appreciated.

qgeissmann, 2022-07-29:
Hi @nwoodweb , 
Thanks for reporting your issue. Each of the three square slots on the side of the arena needs to have a black circle on. we call these 'targets'. You can just print them, and we have a pdf [here](https://www.notion.so/Arenas-c8e5ac0f527e406ab8b82868521ed1a0). This is to automatically find the arena and define the region of interest.

ggilestro, 2022-08-01:
As Quentin said, Arena must have targets. This is explained [in the relevant section of the user manual.](https://www.notion.so/giorgiogilestro/Arenas-c8e5ac0f527e406ab8b82868521ed1a0)

closed 2022-08-01 by ggilestro

nwoodweb, 2022-08-01:
Hello @ggilestro and @qgeissmann 

I wanted to add tracking stickers to the arena before commenting further.

The ethoscopes can now track, and display the tracking interface properly.

However, my second ethoscope is still not working, despite the tracking dots. When tracking is initiated, this specific ethoscope does not display anything, and no longer gives a prompt to stop tracking. In this state the only way to exit this is to login as alarm through ssh and reboot the device. I am still in the process of diagnosing this and Id rather exhaust other options before reflashing  the OS onto the microSD card. 

I apologize for the negligence with regards to the arena module.