gilestrolab/ethoscope #175: “Start Streaming” and “Record Video” on Node webpage not displaying anything

DaveMelkani, 2022-08-08:
When I try to press “Start Streaming” and “Record Video” on the node webpage using the provided ethoscope image on #170, I am unable to see the camera provided. Are there any recommendations on how I should proceed? 
 
Start Streaming:
[image]

Record Video:
[image]

also, if I right-click the loading screen and click open image, I get this:
[image]

ggilestro, 2022-08-09:
Not sure because there is little information but this looks like an issue with poor WIFI connectivity. Can you try by connecting the ethoscope using an ethernet cable?

DaveMelkani, 2022-08-11:
I have tried with an ethernet cable and got the same result. I am wondering if it might be a hardware issue with the actual camera. On the node webpage, it says “picamera : false” so I was wondering if there is a way I can check if the camera can work.

ggilestro, 2022-08-12:
If you sospect a hardware problem, I suggest plugging the ethoscope to a monitor and a keyboard and testing it using the procedure explained [in this page](https://www.notion.so/giorgiogilestro/Software-issues-7de6d3569a014c4eb7f14bb8d5a77d4e).

DaveMelkani, 2022-08-13:
I tried the monitor keyboard set up along with trying out the “raspistill -o testshot.jpg” example, but I think this is only a Raspbian OS feature. I instead download Raspbian OS onto an SD card and tested the camera through there. I got the camera to stream in real time, meaning it is not a hardware issue, but rather a software issue with the ethoscope. I was wondering how I should proceed from this?

pepelisu, 2022-08-15:
How is your network set up, it seems that the ethoscope can be discovered but data cannot be retrieved, I notice that it is number 005, do you have the same problem with the other 4? (in case you have them) 

According to the error, you are running in a 404 error, last_image "Not found" so either, it does not exist on the ethoscope or the ethoscope is not reachable in the port that is needed to get the image.

DaveMelkani, 2022-08-17:
Right now, I am up to ETHOSCOPE_007, and I'm not sure why because I'm just trying to get 1 ethoscope working first. Sometimes the IP address for the node and ethoscope change, resulting in new ETHOSCOPE_00# being added. I was wondering how I can get rid of the old ones, so I go back to ETHOSCOPE_001. 

Also, I'm not sure why this is complicated, as I assume I just need to download the node img from the website and the updated ethoscope img from #170 and everything should be running. 

One thing that I noticed when I was SSH'd into the #170 ethoscope img was that in /tmp, there was no mention of picamera. Also when I would try to run this python script:

```python
from time import sleep
from picamera import PiCamera

camera = PiCamera()
camera.resolution = (1024, 768)
camera.start_preview()
# Camera warm-up time
sleep(2)
camera.capture('foo.jpg')

```
I get an error on this part which makes me think picamera is not properly installed onto the #170 ethoscope img

```python 
 from picamera import PiCamera
```

pepelisu, 2022-08-18:
It would help if you can copy the error here. It could be that the picamera is not installed and therefore giving an error on importing or it can be that the architecture is not matched or somehow the cammera is not available. 

The thing with the name is also not clear to me what is happening, once you have a sd card and the first time you start the ethoscope it should be detected as Ethoscope_000 then you can configure (or set up) the number in the settings dropdown menu. 
You can reuse the numbers. 
Other thing that is not clear to me is the part where you say "IP address for the node and ethoscope change", the IP of the node should not change and by default is 192.168.0.1, if it has another IP address  you should also change that in settings.

ggilestro, 2022-08-18:
I see what the problem is. You're using a beta image that was made for the pi4. This is very experimental and not tested. If you can, please use a pie for the ethoscope and its appropriate image. If all you have is a pi4 then you're in uncharted territory as we never tried to use pi4 as ethoscope.

closed 2022-10-02 by ggilestro